gotthardp/lorawan-server #404: MQTT connector- node red

gmbankar37, 2018-05-30:
Hi ,
I am trying to send the data coming at my gateway to a modbustcp server through mqtt in node-red. I have configured the handler and connector files. My mqtt broker shows disconnected in node-red. Also I have no idea what the handler is exactly doing. I mean do I keep format as ASCII or write my own parse uplink logic. And from where and how exactly is my handler taking data to send to a outside server. 
[image]
[image]
[image]

gotthardp, 2018-05-30:
It depends on what your device is sending. If it's sending ASCII Text, set the Payload to ASCII and leave the Parse Uplink empty. If it's sending some binary structure that you want to decode, leave Payload empty and set some Parse Uplink function.
https://github.com/gotthardp/lorawan-server/blob/master/doc/Handlers.md#administration

gmbankar37, 2018-05-30:
> doc/Handlers.md#administration

For my external application to be working do I have to disable my internal
application?

gotthardp, 2018-05-30:
As long as you assign different names to the different applications all can be enabled.

closed 2018-06-08 by gotthardp