gotthardp/lorawan-server #407: HTTP Connector SSL cihper suite configuration

pmst, 2018-06-01:
I have configured the HTTP connector to point to a server using ssl (https). The remote system (frontend server is caddy) is refusing connection due to the use of a non accepted cipher suite. The error is the following:
> {gun_error,<0.385.0>,{stop,{goaway,0,inadequate_security,<<"Prohibited TLS 1.2 Cipher Suite: c014">>},'Client is going away.'}}

Is there a way to configure the client library cipher suites to use ?

gotthardp, 2018-06-04:
Configuring individual ciphers is not possible and also not recommended.
What Erlang version are you using? Do you use the pre-compiled server version, or you build your own?

pmst, 2018-06-07:
I was more thinking of removing a deprecated cipher. My erlang is installed from the package mentioned in the Installation instructions: erlang-solutions_1.0_all.deb, current version is 20.3

gotthardp, 2018-08-25:
I implemented a new config option `ssl_options` for 0.6.0

closed 2018-08-25 by gotthardp