gotthardp/lorawan-server #408: Remove and duplication of server

gmbankar37, 2018-06-01:
How can I remove the lorawan-server if its implemented on a Raspberry pi? 
Also If I want to duplicate the server on a different host do I have to just change the Mnesia contents or is there any other method?

gotthardp, 2018-06-01:
I am not sure what do you mean. The server stores everything in the Mnesia database. If you stop the server you can duplicate or delete the database as you wish,

urbie-mk2, 2018-06-02:
My question is of similar nature: 
Can I create cluster of lorawan-servers (think packet forwarder and lorawan-server on the same device) while each cluster node uses the same database (which is mnesia if I am not mistaken), so that when more than one gateway receives the uplink payload of an end device only one lorawan-server would handle the payload (and not two or more lorawan-servers) ?

altishchenko, 2018-06-03:
@urbie-mk2 Database clusterisation can be done with a little black magic using Mnesia features, so two servers can use the same replicated database. I did one project, where the database actually belonged to a third application and servers were replicating configuration from it. As for multi-server deduplication, this IMHO can not be done if I got your problem right. One server deduplicates between different gateways based on timing and quality parameters, but servers will not talk to each other, only replicate mnesia data.

urbie-mk2, 2018-06-03:
Thanks for the idea with database clusterisation.

 I am searching for different ways for make the lorawan network, including its components more reliable , or highly available without having a single point of failure while at the same time not requiring additional hosts for making this possible (neither real devices nor virtualized or containerized). The good thing is that no support for device roaming is required.

It is less of a problem if a gateway fails but the network server is way more important. If it fails, the whole network is down. So the idea behind the cluster is to avoid this problem.

In this case I will need to consider changing the lorawan-server in a way that would allow it to operate as a cluster.

altishchenko, 2018-06-03:
@urbie-mk2 I should warn you ahead :) a little bit. The main framework with the server is Erlang/OTP, meaning that if one of the server's components fails for some reason it is just restarted appropriately in no time - the server stays alive until the hardware underneath it is alive.
Planning for HA and clusterization is a good thing, but it is quite complicated and there is no 'one size fits all'. And now the truth: If you want a message from the gateway to be received 100%, then you should plan for: a bunch of stateless **load balancers**, that will disperse the traffic around; a bunch of physical hosts running the network server instances, behind those balancers; then situ continues when you step up to your application - that also should be behind the balancers and duplicated... Sounds paranoid? It is!
To the main story - what do you _really_ need? Or, to make it easier, what are you trying to achieve? What is your **business case**?

urbie-mk2, 2018-06-03:
@altishchenko
That should be no problem. I just had to know if there was already a mechanism in loraserver that could be used for HA.

Gateways are being installed redundantly up to three in near proximity so I thought of using the machines for clustering and high availability. I always prefer a software system that handles itself without requiring manual intervention.

But I am considering different approaches now since I understand that there is not one solution that works for everybody.

It may not fail is what I am told, so that is what I am going to deliver. I could ask more questions to have a clearer picture but they get confused sometimes when I use too much technical information.

The software under development is a product that is part of a bigger software system which is a customizable product.

urbie-mk2, 2018-06-04:
@gmbankar37 
Given the fundamental idea of a lorawan  gateway, it is only supposed to host the packet forwarder together with utility programs (think network manager, firewall, etc) as a place for the handover from the lorawan phy layer to the ethernet phy layer and vice versa which doesnt mean you cannot put more software (e.g. lorawan-server) on the gateway, too. 

On huge lorawan networks it is simply not possible to put a lorawan server on the gateway , and given the fact that a lorawan itself can further be divided into different servers (networkserver (forwarding, servicing, homing), join server, app server) it would not make sense to do so but for small applications of only a handful of gateways within a secure tamper-proof environment it is not unthinkable to run lorawan-servers on the gateway itself. I haven't seen any gateway manufacturer who does it though.

gotthardp, 2018-06-26:
I will provide some guidelines/suggestion for clustering. See Issue #422.

closed 2018-06-26 by gotthardp