gotthardp/lorawan-server #409: some pages not found on 0.5.4

hallard, 2018-06-03:
I upgraded from 0.5.3 to 0.5.4 on my PiZero and I noticed something strange, even if the server works fine, I got 2 menus that fire a page not found, the others menus are working fine
the menu are "the main page" and "events" 

[image]

the other (main page) is this one /admin#/dashboard

Anyone noticed this or it's just my config that have an issue ?

gotthardp, 2018-06-04:
I experience the "Not Found" error as a temporary state after upgrade while the web-admin was running. The error goes away in about 1 minute. Did your error persist?

hallard, 2018-06-04:
Well, yes it persist, I think it's more on syntax than on update.
On Publish events from WS connector I changed `/{app}/events` to `ws/{app}/events` and it's gone. looks like expanding the 1st one {app} at 1st level give some trouble (but only on event, works fine on uplink)

gotthardp, 2018-06-04:
yeah, i know what happended: the `{app}/events` got confused with `api/events`. I made a fix for that. Thanks for reporting this!

gotthardp, 2018-06-05:
Will be fixed in v0.5.5

closed 2018-06-05 by gotthardp