gotthardp/lorawan-server #411: Eating up Free Memory

netuserlv, 2018-06-05:
At the beginning I would like to thank the maker of this server, it is really grate so far and all. But I was wondering am I doing something wrong because the server is showing less and less available ram memory? Can there be something done about that, does it save something on it and how can i delete it. Because I am forwarding everything to a node red which is saving the data somewhere else and I kinda don`t know what to do.... 
[image]

gotthardp, 2018-06-05:
Hello. Thank you! The memory can be consumed by a caching mechanism, so it may eventually get released again. What Erlang version are you using? Could you try upgrading Erlang to see if something changes?

cristianhumelnicu, 2018-08-02:
Hi Petr ,
I've upgrade erlang and system still eating a lot of memory and that is critical when someone (like me) have lorawan server installed on RPI3

Thx !

gotthardp, 2018-08-02:
@cristianhumelnicu would you be able to monitor the consumed memory over time (days) and see how it evolves?

xypron, 2018-11-29:
There was a display issue for free memory that has been fixed by merging #528. So I think this issue can be closed now.

gotthardp, 2018-11-30:
Right. Thanks!

closed 2018-11-30 by gotthardp