gotthardp/lorawan-server #413: Openembedded lorawan-server installation issue

rkdm91, 2018-06-13:
Hello,

I am facing an issue trying to add lorawan-server in my Openembedded project. I have added meta-erlang and meta-lorawan to bblayers.conf, and have added lorawan-server in my local.conf.

Please find the following bitbake log generating by the error:

ERROR: lorawan-server-0.5.5-r0 do_compile: oe_runmake failed
ERROR: lorawan-server-0.5.5-r0 do_compile: Function failed: do_compile (log file is located at /home/rkdm/oe-core-v2/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/lorawan-server/0.5.5-r0/temp/log.do_compile.2509)
ERROR: Logfile of failure stored in: /home/rkdm/oe-core-v2/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/lorawan-server/0.5.5-r0/temp/log.do_compile.2509
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4 release
| /usr/bin/env: ‘escript’: No such file or directory
| Makefile:35: recipe for target 'release' failed
| make: *** [release] Error 127
| ERROR: oe_runmake failed
| WARNING: /home/rkdm/oe-core-v2/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/lorawan-server/0.5.5-r0/temp/run.do_compile.2509:1 exit 1 from 'exit 1'
| ERROR: Function failed: do_compile (log file is located at /home/rkdm/oe-core-v2/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/lorawan-server/0.5.5-r0/temp/log.do_compile.2509)

I may have miss something (package installation/configuration). Did you face this type of issue before ?

gotthardp, 2018-06-13:
It is the same as #349. The problem is that the `escript` is missing. We didn't find the reason why it was missing.

rkdm91, 2018-06-13:
Thank you for your answer,it seems it link to the building environment, is it related to ubuntu ? did you try this with ubuntu or another distribution ?

gotthardp, 2018-06-13:
Good catch. I am not using Ubuntu. I did use Debian, so if you can switch distributions it may be the quickest way forward.

altishchenko, 2018-06-14:
@rkdm91 I built the server on Ubuntu successfully, though I didn't use bitbake.
What is the output of `which escript` on your machine?
Did you try building the server by hand, just in case?

rkdm91, 2018-06-14:
All files seems to be in place and installed : 
$ which escript
/usr/bin/escript

I didn't try to build the server by hand, i will try it.

gotthardp, 2018-06-14:
The problem is that `which escript` tells there is escript on the machine, but I think that during the build the paths are redirected to the build environment (the erlang-native thing).

gotthardp, 2018-06-26:
Any update? Was the issue solved?

rkdm91, 2019-02-19:
HI,
Sorry for the late answer, I didn't solve the issue.
I Just install erlang with its modules from meta-erlang and use the Binary Release for linux.