gotthardp/lorawan-server #414: Server update

jot4p, 2018-06-14:
Hi, 

I have the version 0.53.. and I need to update for 0.55. 

It's impossible to copy db (system crash) ..

I have >100 nodes ...  I need to create the devices, handlers, connectos, ADR settings  ?

gotthardp, 2018-06-14:
What db copy you mean and what db crash you experienced?

gotthardp, 2018-06-26:
Any update? Shall I close this issue?

gotthardp, 2018-06-26:
And what error do you get?

jot4p, 2018-06-27:
Ok my problem is based in the problem of my computer name.. I have 1 lorawan in server and other in my personal computer.. but your database is based on computer name. 

I'm trying to create "lorawan-server-beta" in my server but I have this:

> Protocol 'inet_tcp': the name lorawan@ns373704 seems to be in use by another Erlang node

It's possible to use allways the same name ? or control this ?

gotthardp, 2018-06-27:
The node name is based on the hostname. If you set identical hostnames you get identical database files, but the best (and recommended) way to backup is to use the scripts I provided.

jot4p, 2018-06-27:
Can you provide the URL ?

gotthardp, 2018-06-27:
See dbexport and dbimport in https://github.com/gotthardp/lorawan-server/tree/master/scripts

jot4p, 2018-06-27:
Do you have the same for windows ?

gotthardp, 2018-06-27:
No.

gotthardp, 2018-08-01:
The import/export scripts for windows will be in 0.6

closed 2018-08-01 by gotthardp