gotthardp/lorawan-server #415: Many missed Uplinks

gmbankar37, 2018-06-21:
Hello,  
I have 5 devices connected to the gateway and each are sending packets at the same time. I using the handler as application and have mqtt connector to give data to another server. When I connect 1 device I dont get any missed uplinks but when the no. of devices increase I get many missed uplinks. This is a problem for me. What should I change? Also are missed uplinks re-transmitted? Thank you.

gotthardp, 2018-06-21:
If more devices transmit at the same time the transmission may be destroyed by interference. There is no way out other than to reschedule the uplinks to not overlap. The uplinks will get retransmitted if you turn on the confirmed mode.But this will only increase the number of messages and make the problem worse.

gmbankar37, 2018-06-22:
And what is the importance of Adaptive Data rate? If there are 8 channels
it shouldnt be problem right?

gotthardp, 2018-06-22:
If the devices actually use all 8 channels then the probability of a collision is lower. Note that for some SF/DR the list of frequencies may be lower than 8, at least in europe.

gotthardp, 2019-03-16:
Closing inactive discussion

closed 2019-03-16 by gotthardp