gotthardp/lorawan-server #416: Loraserver do not work with erlang 21

cristianhumelnicu, 2018-06-22:
Hi Petr,

I've upgrade erlang to 21.release edition and after that loraserver do no start 

I0ve remove after and loraserver and reinstall it but still no luck 

That is in debug log

2018-06-22 11:12:58.135 [warning] <0.412.0>@lorawan_connector_mqtt:terminate:252 Connector TB-RPI terminated: {{badmatch,{error,{badarg,[{erlang,apply,[{gen_logger,console_logger,4},warning,[]],[]},{emqttc,init,1,[{file,"/root/lorawan-server/_build/default/lib/emqttc/src/emqttc.erl"},{line,397}]},{gen_fsm,init_it,6,[{file,"gen_fsm.erl"},{line,351}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,249}]}]}}},[{lorawan_connector_mqtt,connect,3,[{file,"/root/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_mqtt.erl"},{line,120}]},{lorawan_connector_mqtt,'-execute_hierarchy_updates/3-fun-0-',4,[{file,"/root/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_mqtt.erl"},{line,102}]},{lists,filtermap,2,[{file,"lists.erl"},{line,1317}]},{lorawan_connector_mqtt,execute_hierarchy_updates,3,[{file,"/root/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_mqtt.erl"},{line,97}]},{lorawan_connector_mqtt,handle_info,2,[{file,"/root/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_mqtt.erl"},{line,162}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,637}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,711}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,249}]}]}

Any ideea ?

Thank you !

gotthardp, 2018-06-22:
The Erlang 21 may be incompatible with some of the libraries currently used. It may take some time until all the dependencies are fixed.

cristianhumelnicu, 2018-06-22:
Then i will reinstall all , Erlang 19.1 and after loraserver .

BTW i did try to instal erlang21 because loraserver seems to eating all RAM memory  ... a lot alarm regarding that > 

Thnak you Petr!

gotthardp, 2018-06-22:
You can try Erlang 20 or 19.3. I know there were some memory problems before 19.2 or 19.3.

cristianhumelnicu, 2018-06-22:
Hi again 
I have re install OTP-19 and loraserver on RPI3 with raspibian , guided by installing instruction from here.
gw@gateway:~ $ erl
Erlang/OTP 19 [erts-8.2.1] [source] [smp:4:4] [async-threads:10] [kernel-poll:false]
In this moment loraserver is working but i still have the problem with RAM Memory

Coul you tell me if is posible to upgrade erlang to 2.0 and how ?

THX!

gotthardp, 2018-06-22:
No worries. Upgrade to Erlang 20 shall work. Try to follow the guidelines here:
https://www.erlang-solutions.com/resources/download.html

gotthardp, 2018-06-26:
The incompatibility with Erlang 21 is caused by https://github.com/emqtt/gen_logger/issues/5

gotthardp, 2018-08-23:
Solved. (By a workaround for now.)

closed 2018-08-23 by gotthardp