gotthardp/lorawan-server #423: Integration with IBM Watson Iot

myfreescalewebpage, 2018-06-26:
Hi,

First of all thanks for your great job with this server!

I have followed integration guide to use Watson IoT (https://github.com/gotthardp/lorawan-server/blob/master/doc/Integration.md#ibm-watson-iot-platform) but I have a small difficulty to achieve it.

I have created a free account on IBM BlueMix and created a Watson IoT service. I have created a server device type (called "EaConnect") and a server device, retrieved the API key and followed your instructions.
Actually the lorawan server is displayed "Connected" on watson IoT but impossible to see my LoRaWAN devices on the plateform ! I have created a device type (called "Watteco") and a device (but I have understand it was not necessary, right?) and it doesn't solve the issue.
Here are some screenshots of my connector configuration:

[image]

[image]

And a screenshot of my Watson IoT device view, show the lorawan server connected but not my device:

[image]

i have performed a dump of the connection attached below. The JOIN and the data frames sent by the LoRaWAN device seems to be forwarded to watson iot.

[mqtt-join.zip](https://github.com/gotthardp/lorawan-server/files/2137567/mqtt-join.zip)

Any help is appreciated to understand what is not correctly configured :-)

Thanks,
Joel

gotthardp, 2018-06-27:
The devices will be created automatically (by Watson) once they send some data. Could this be the case?

myfreescalewebpage, 2018-06-28:
Hello
Sorry I don't understand what you mean when you say "could this be the case" ?
Do you think the tcp dump is correct ? Do you have a reference one ?
I don't know if the issue is my configuration of the lorawan server or the watson side...
Thanks,
Joel

myfreescalewebpage, 2018-06-28:
I have just deleted my LoRaWAN device on watson iot, and restarted it. The device has been created automatically when join and data have been received from the lorawan server. Seems communication is ok, but how to get device displayed "connected" on watson iot and how to see data sent ?
Thanks,

myfreescalewebpage, 2018-06-28:
Finally solved, but I don't know how I have done it. Just played a little bit with watson iot.
Your server is perfect and the integration guide is ok.
Thanks again for your time and good luck for the next step in this project !

closed 2018-06-28 by myfreescalewebpage

gotthardp, 2018-06-28:
Thanks for letting me know and sorry that I couldn't offer more help. My experiences with IBM Watson are not very big.