gotthardp/lorawan-server #424: Modificar lorawan-server el campo 'rxframe'

thegamecena619, 2018-06-26:
Hello,

Maybe you can tell me how I can do to create a field similar to this one:

    var rxframes = nga.entity ('rxframes')
        .identifier (nga.field ('frid'))
        .readOnly ();

The objective is to duplicate the 'rxframe' field, that is, obtain a 'rxframe2' and use the variable:

nga.field ('data')
            .template (function (entry) {
                if (entry.values.data)

How could I do it? Thank you very much in advance.

gotthardp, 2018-06-26:
And why do you do that?

thegamecena619, 2018-06-26:
is that I need to do a job that modification and I need to have two fields with the same characteristics.

gotthardp, 2018-06-26:
I still don't understand why do you want to do that modification.

thegamecena619, 2018-06-26:
[image]

The idea is to have two fields 'application' and 'application 1'

gotthardp, 2018-06-26:
I modified https://github.com/gotthardp/lorawan-server-demoapp to show the right way how to do what you are trying to achieve.

closed 2018-07-16 by gotthardp