gotthardp/lorawan-server #425: batch script for windows update ( Import and export )

jot4p, 2018-06-28:
Hi, 

I convert your update files for windows:

create file: dbexport_windows.bat
```
REM echo OFF

for /f "tokens=1-3 delims=- " %%a in ('date/t') do (
set dd=%%c
set mm=%%b
set yy=%%a
)

set SERVER=localhost:8080
set AUTH=--user=admin --password=admin
set DIR=backup-%yy%-%mm%-%dd%
mkdir %DIR%

for %%G in (users,gateways,networks,multicast_channels,profiles,devices,nodes,ignored_nodes,handlers,connectors) DO (
	echo Processing db %%G
	wget -nv %AUTH% --output-document=%DIR%/%%G.json http://%SERVER%/api/%%G
)
```
create file: dbimport_windows.bat

```
REM echo OFF

for /f "tokens=1-3 delims=/ " %%a in ('date/t') do (
set dd=%%a
set mm=%%b
set yy=%%c
)

set SERVER=localhost:8080
set AUTH=--user=admin --password=admin
set DIR=%1

echo LIST FILES:
FOR %%L in (%DIR%\*.json) DO (
	echo %%L
)

echo POST FILES:
FOR %%F in (%DIR%\*.json) DO (
 wget -nv %AUTH% --header=Content-Type:application/json --post-file=%%F http://%SERVER%/api/%%~nF
)
```

I test for update 0.53 to 0.55 with success

gotthardp, 2018-06-28:
Cool!!!! Thanks a lot!

gotthardp, 2018-07-16:
Just added them to the master branch. Will be on 0.6.0.

closed 2018-07-16 by gotthardp