gotthardp/lorawan-server #426: How to re-build from source to adapt the NOT_SEMTECH_MOTE problem?

kcyung3663, 2018-07-01:
Currently I am using Cisco gateway to connect with this server.

It was all fine for either gateway connection/ device joinAccept.

However, when it comes to receiving rxpk from devices(through gateway), not_semtech_mote is displayed, as show below.
[image]

It is known that the rxpk format outputted is like this:
```
{  
   "rxpk":[  
      {  
         "jver":2,
         "tmst":23479876,
         "brd":0,
         "aesk":0,
         "freq":923.400000,
         "stat":1,
         "modu":"LORA",
         "datr":"SF10BW125",
         "codr":"4/5",
         "size":23,
         "data":"AAEBAQEBAQEBAQEBAQEBAQGj7aQR08Y=",
         "rsig":[  
            {  
               "ant":0,
               "chan":1,
               "rssic":-103,
               "lsnr":14.0,
               "ftstat":0,
               "rssis":-103,
               "rssisd":0,
               "foff":841
            },
            {  
               "ant":1,
               "chan":17,
               "rssic":-102,
               "lsnr":14.0,
               "ftstat":0,
               "rssis":-103,
               "rssisd":0,
               "foff":841
            }
         ]
      }
   ]
}
```
Item likes `time`, `rfch` are missing and `chan`,`rssi`,`lsnr` are stored into a sub-level under the property `rsig`.

As the packet forwarder of the gateway is unmodifiable, I am thinking about that the only way is to change the rules playing on server.

Hence, I am hoping if anyone could give me some insights.

Thanks.

gotthardp, 2018-07-01:
Hello. Support for the `rsig` thing is implemented since 0.5.4. The issue you are experiencing is not related to the gateway, but to the device/application. You set the application to Semtech Mote, but your device is not sending application payload in compliance with what Semtech Mote (was) using.
What device you have? Perhaps you need to define your own application Handler.

kcyung3663, 2018-07-01:
Hi Gotthardp.

For the device, please see the image attached.
[image]

I don't quite understand what do you mean by payload compliance. But for Semtech Mote, in our understanding, it is talking about the structure of rxpk, isn't it?

And to define an application handler, is it just to modify the `lorawan_handler.erl` and re-make from the source?

gotthardp, 2018-07-01:
Yeah, that is NOT a Semtech mote. You don't have to rebuild anything, but you need to define the payload structure. Have a look at https://github.com/gotthardp/lorawan-server/blob/master/doc/Handlers.md

Apaisal, 2018-07-02:
Or remake new your application mote from
src/lorawan_application_semtech_mote.erl template file. I made my
application mote from this file.

Regards,
Anol

kcyung3663, 2018-07-02:
Thanks both @gotthardp and @Apaisal .
I am now trying to do something with handler setting, however, `parse_failed` always show.

Regarding the rxpk format aforementioned, I tried setting the `Parse Uplink` as either of the followings:
```
fun(Fields, <<Jver, Tmst, Brd, Aesk, Freq, Stat, Modu, Datr, Codr, Size, Data, Rsig>>) ->
  Fields#{jver => Jver, tmst => Timestamp, brd => Board, aesk => AesKey, freq => Freq, stat => Stat, modu => Module, datr => Datarate, codr => Coderate, size => Size, data => Data, rsig => Rsig}
end.
```
```
fun(_Port, <<Data:128>>) ->
  #{data => binary_to_list(Data)}
end.
```
Other fields are set to blank as downlink message is not required at the current stage.

In my understanding, the function is to directly map object properties into JSON format. But given that the incoming packet is already in JSON format, what actually does it do further?
**The gateway-NS data grep by WireShark:**
[image]

Thanks.

gotthardp, 2018-07-02:
The `rxpk` JSON contains a data field. The server does parse that `data` field and extracts custom application data sent by your device. These data are always binary.

Your fun() is syntactically incorrect, hence the error you're getting.

First thing to do is to verify WHAT does your device send.

kcyung3663, 2018-07-02:
```
AppData.Buff[i++] = AppLedStateOn;
  AppData.Buff[i++] = ( pressure >> 8 ) & 0xFF;
  AppData.Buff[i++] = pressure & 0xFF;
  AppData.Buff[i++] = ( temperature >> 8 ) & 0xFF;
  AppData.Buff[i++] = temperature & 0xFF;
  AppData.Buff[i++] = ( humidity >> 8 ) & 0xFF;
  AppData.Buff[i++] = humidity & 0xFF;
  AppData.Buff[i++] = batteryLevel;
  AppData.Buff[i++] = ( latitude >> 16 ) & 0xFF;
  AppData.Buff[i++] = ( latitude >> 8 ) & 0xFF;
  AppData.Buff[i++] = latitude & 0xFF;
  AppData.Buff[i++] = ( longitude >> 16 ) & 0xFF;
  AppData.Buff[i++] = ( longitude >> 8 ) & 0xFF;
  AppData.Buff[i++] = longitude & 0xFF;
  AppData.Buff[i++] = ( altitudeGps >> 8 ) & 0xFF;
  AppData.Buff[i++] = altitudeGps & 0xFF;
```
This is what placed inside the core of device.

gotthardp, 2018-07-02:
Then try something like
```erlang
fun(Fields, <<Press:16, Temp:16, Hum:16, Batt, Lat:24, Lon:24, AltGps:16>>) ->
  Fields#{press => Press, temp => Temp, hum => Hum, batt => Batt, lat => Lat, lon => Lon, alt => AltGps}
end.
```
The first line does parse the binary data as created by the device and the second line does put these values into variables.

kcyung3663, 2018-07-02:
Tried, but the outcome is still `parse_failed`.

We tried using both lora device and lora network test terminal(an object like the traditional phone), same result.

One thing worths mentioning is, by putting the payLoad into lora packet decoder (installed locally, similar to https://lorawan-packet-decoder-0ta6puiniaut.runkit.sh/), the outcome is just something unreadable.

```
Message Type = Data
            PHYPayload = 800100C0CC8002002181074F07C065840B3E8B49C62D200D77A413FBF4C69FD319DA3A74B70DB3

          ( PHYPayload = MHDR[1] | MACPayload[..] | MIC[4] )
                  MHDR = 80
            MACPayload = 0100C0CC8002002181074F07C065840B3E8B49C62D200D77A413FBF4C69FD319DA3A
                   MIC = 74B70DB3

          ( MACPayload = FHDR | FPort | FRMPayload )
                  FHDR = 0100C0CC800200
                 FPort = 21
            FRMPayload = 81074F07C065840B3E8B49C62D200D77A413FBF4C69FD319DA3A

                ( FHDR = DevAddr[4] | FCtrl[1] | FCnt[2] | FOpts[0..15] )
               DevAddr = CCC00001 (Big Endian)
                 FCtrl = 80
                  FCnt = 0002 (Big Endian)
                 FOpts =

          Message Type = Confirmed Data Up
             Direction = up
                  FCnt = 2
             FCtrl.ACK = false
             FCtrl.ADR = true

MIC: OK
Payload: ��<���F|�0��Z
```

gotthardp, 2018-07-02:
Well, your FRMPayload is 26 bytes but in your `AppData.Buff[i++]` there is only 15 bytes. I believe your device is sending something else.
For a start, leave both "Parse" fields empty and add `data` to your "Uplink Fields".

kcyung3663, 2018-07-02:
In that case, with only `data` in Uplink Fields, no other errors were prompted.
(Starting from 22:09)
[image]

closed 2018-07-05 by gotthardp

altishchenko, 2018-07-06:
@kcyung3663 Sorry for the late question, but can you tell me what type of Cisco gateway do you use? The standard ones (IXMs) normally come with the Actility packet forwarder, which is TCP based and the only one available from the Cisco's web-site. How did you manage to make it 'talk' semtech packet forwarder?

kcyung3663, 2018-07-09:
@altishchenko The gateway model is IXM-LPWA-900-16-K9.

We didn't use the Acility but the original one.

kcyung3663, 2018-07-11:
Would you leave me your whatsapp/ telegram?

altishchenko, 2018-07-12:
@kcyung3663 sure! +7-916-704-6157