gotthardp/lorawan-server #427: When specifying downlink port, no txpk is sent and debug.log shows error in lorawan_mac.erl (line 472)

cmenscher, 2018-07-03:
We are attempting to send downlinks to a device using FPort 7.  We are using the HTTP connector to instruct lorawan-server to transmit a downlink, which seems to work unless we specify a port number.

When using the following curl command...

```
curl -v --digest -u admin:[OUR PASSWORD] -H "Content-Type: application/json" -X POST http://[OUR URL]:8080/down/00137A1000000993 -d "{\"data\":\"0EFF000000000000000000\"}"
```

lora_pkt_fwd running on our MultiTech Conduit gateway reports the following...

```
JSON up: {"rxpk":[{"tmst":1070306796,"chan":3,"rfch":0,"freq":902.900000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":7.8,"rssi":-94,"size":24,"data":"QMbJNQCATwEGdm/7O9dAhGrQoCdHPoHH"}]}
INFO: [up] PUSH_ACK received in 10 ms
INFO: [down] PULL_RESP received  - token[215:49] :)

JSON down: {"txpk":{"powe":11,"imme":false,"tmst":1071306796,"codr":"4/5","datr":"SF10BW500","freq":925.1,"modu":"LORA","rfch":0,"ipol":true,"size":24,"data":"YMbJNQCAGAAGrphEmqKjBZatWbxCclzt"}}
```

However if we use the following command, which includes the port number...

```
curl -v --digest -u admin:[OUR PASSWORD] -H "Content-Type: application/json" -X POST http://[OUR URL]:8080/down/00137A1000000993 -d "{\"port\":\"7\", \"data\":\"0EFF000000000000000000\"}"
```

We only see an rxpk in the packet forwarder log, and never a txpk.  In both cases, we have checked the node's page in lorawan-server and can see the downlink waiting to be sent.  In the first curl example, the "Port" column is blank in the Downlinks section.  In the second, the "Port" column says "7", as intended.  In both cases, after an uplink is sent from the device the downlink disappears from the page, which indicates lorawan-server thinks a downlink txpk was sent.  But in the case where a port is specified, the gateway never receives a txpk, nor generates an error of any kind.  However, 
 watching lorawan-server debug.log as we transmit the uplink we can see it output the following:

```
2018-07-03 16:41:21.367 [error] <0.12880.19>@lorawan_mac:encode_frame:472 gen_statem <0.12880.19> in state uplink terminated with reason: bad argument in lorawan_mac:encode_frame/8 line 472
2018-07-03 16:41:21.367 [error] <0.12880.19>@lorawan_mac:encode_frame:472 CRASH REPORT Process <0.12880.19> with 0 neighbours crashed with reason: bad argument in lorawan_mac:encode_frame/8 line 472
2018-07-03 16:41:21.367 [error] <0.391.0>@lorawan_mac:encode_frame:472 Supervisor lorawan_handler_sup had child handler started with lorawan_handler:start_link() at <0.12880.19> exit with reason bad argument in lorawan_mac:encode_frame/8 line 472 in context child_terminated
```

Presumably, this is a bug, and is why the downlink is never sent.  Unfortunately, this is a major issue for us as we have to be able to target a port for sending these particular downlinks, as specified by a manufacturer.

gotthardp, 2018-07-03:
You are sending the number as a string (in quotation marks). Could you try sending it as a number, please?
```
curl -v --digest -u admin:[OUR PASSWORD] -H "Content-Type: application/json" -X POST http://[OUR URL]:8080/down/00137A1000000993 -d "{\"port\":7, \"data\":\"0EFF000000000000000000\"}"
```

cmenscher, 2018-07-03:
Hey!  That worked. 😎 Thanks for the insanely fast response.

closed 2018-07-03 by cmenscher