gotthardp/lorawan-server #428: Getting data thru connector only once after creating object

nupik, 2018-07-04:
**1. When I create Node(OTAA) first message is arrived thru connector(tested on /admin/ws.html)**
04.07.2018 13:26:16 RECEIVED: {"appargs":"t1","data":"F100E801DE2600000000","datetime":"2018-07-04T10:26:17Z","devaddr":"015F034C","humidity":48.8,"pressure":995,"temp":24.1}

**2. Second message not received (in debug log only "retransmitting") - what it is?**
2018-07-04 13:25:55.590 [debug] <0.502.0>@lorawan_connector_ws:websocket_init:78 WebSocket connector <<"
2018-07-04 13:26:17.826 [debug] <0.506.0>@lorawan_mac:ensure_rxwin:325 <<"015F034C">> RXWindow initializ
2018-07-04 13:26:17.826 [debug] <0.506.0>@lorawan_mac:ensure_adr:315 <<"015F034C">> ADR initialized
2018-07-04 13:26:18.030 [debug] <0.506.0>@lorawan_mac_commands:store_actual_adr:134 ADR indicator set to
2018-07-04 13:26:18.031 [debug] <0.506.0>@lorawan_mac_commands:build_fopts:73 015F034C <- [dev_status_re
2018-07-04 13:26:53.867 [debug] <0.508.0>@lorawan_handler:retransmit:210 015F034C retransmitting
2018-07-04 13:27:22.297 [debug] <0.578.0>@lorawan_handler:retransmit:210 015F034C retransmitting
2018-07-04 13:27:28.908 [debug] <0.502.0>@lorawan_connector_ws:terminate:143 WebSocket terminated: remot
2018-07-04 13:27:29.414 [debug] <0.595.0>@lorawan_connector_ws:websocket_init:78 WebSocket connector <<"
2018-07-04 13:29:00.217 [debug] <0.615.0>@lorawan_handler:retransmit:210 015F034C retransmitting

gotthardp, 2018-07-05:
Reatransmit means that either the server received uplink with the same FCnt, or the device didn't confirm a confirmed downlink and application asked for a retransmission.

nupik, 2018-07-05:
Yes, this device work without confirmation, how I can handle in this case?

gotthardp, 2018-07-05:
Could you have a look on the Received Frames and confirm the device keeps sending the same FCnt?

nupik, 2018-07-05:
Yes, FCnt is "0" :(

Received | Application | DevAddr | MAC | U/L RSSI | U/L SNR | FCnt | Confirm | Port | Data
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
2018-07-05 12:48:01 | WebSocketRad | 015F034C | 000094E36DD21D00 | -117 | 2.8 | 0 |   | 1 | 0701E0011D2700000000
2018-07-05 12:47:45 | WebSocketRad | 015F034C | 000094E36DD21D00 | -114 | 4.5 | 0 |   | 1 | 0701E7011D2700000000

gotthardp, 2018-07-05:
This explains the retransmissions. You need to figure out why the device does not increase it. Also the payload looks the same-- could the device be waiting for some downlink?

nupik, 2018-07-05:
This is new devise, just in development state, what is why FCnt not increase. Now I send request to developer for fixing it. But in common case, as I understand it is necessary to increase this counter.
Thanks for support.

gotthardp, 2018-07-05:
Yes, it is. :)

closed 2018-07-05 by gotthardp