gotthardp/lorawan-server #430: Integration with ThingsBoard - No Telemetry data

myfreescalewebpage, 2018-07-08:
Hello,
I'm currently trying to integrate with ThingsBoard. I have installed ThingsBoard, created a device, set up LoRaWAN server with the right parameters indicated at https://github.com/gotthardp/lorawan-server/blob/master/doc/Integration.md#thingsboard-open-source-iot-platform.
Attributes tab of ThingsBoard works correctly, join event is displayed when the LoRaWAN device joins and test event is displayed when I click the Test button in the Handler configuration.
However, I have no telemetry data at all when the LoRaWAN device sends data. Parse uplink is working, I have no error displayed on the web UI.
I have followed https://thingsboard.io/docs/getting-started-guides/helloworld/#pushing-data-from-the-device to test the ThingsBoard installation and configuration, it works correctly.
Just seems the LoRaWAN server is not sending telemetry data...
Any ideas ?
I have found a similar issue : https://github.com/gotthardp/lorawan-server/issues/388 but seems to be corrected (I'm using server V0.5.5)
Thanks for your help
Joel

myfreescalewebpage, 2018-07-08:
Just understood.
The telemetry doesn't support 'complex' object at all. In the Integration guide when you say 'Make sure that your Handler Parse Uplink and Parse Event produce a flat list of string, boolean and numeric fields only.' please can you detail that 'Uplink Fields' should be left empty because it can contain arrays and objects ('allgw' for example).
I have just logged the issue on Thingsboard github (https://github.com/thingsboard/thingsboard/issues/923) to know if support is planned.
Closing this issue.
Joel

gotthardp, 2018-07-09:
What kind of data you'd need to see in Thingsboard? I can try "flattening" them.

gotthardp, 2018-07-13:
And do you really want to see RSSI from _all_ gateways? The server finds the "best" gateway (with highest RSSI) and puts the RSSI and MAC into the flat list.

gotthardp, 2018-07-16:
Good, thanks.

closed 2018-07-16 by gotthardp