gotthardp/lorawan-server #431: unable to post to MongoDB using Conenctor

kcyung3663, 2018-07-09:
Hi all.

I've include some images down below to tell the situation.

Connector setting:
[image]

MongoDB:
[image]
[image]

Received frames:
[image]

An additional question for @gotthardp, regarding the setting for the uplink parse field in #426, we are still unable to solved the parse_failed error in order to map the 28byte (or 32byte for testing) data into JSON format.

Here is the handler setting:
[image]

altishchenko, 2018-07-11:
@kcyung3663 

What bothers me here is the **Payload: ASCII**..
From the .doc:

> ASCII Text
> The payload will get stored into the `text` field as ASII characters.

But from the 'Received Frames' the payload is '00278109...' in hex and this is obviously not clear text ASCII judging by the first 00. Try setting the payload type to `undefined`, add `data` to `Uplink Fields`  and see what happens.

Heiner92, 2018-07-31:
I have the same problem. I receive the payload from the gateway and the server is trying to send the data via my connector "Dumper" to a mongoDB on the local network.

Here is the error log I get:

> 2018-07-31 17:32:12.393 [debug] <0.304.0>@lorawan_connector_sup:stop_child:23 Stop {mongodb,<<"Dumper">>}
2018-07-31 17:32:12.393 [debug] <0.7357.1>@lorawan_connector_mongodb:log_termination:79 Connector Dumper terminated: shutdown
2018-07-31 17:32:12.393 [debug] <0.304.0>@lorawan_connector_sup:start_child:16 Start {mongodb,<<"Dumper">>}
2018-07-31 17:32:12.394 [debug] <0.7365.1>@lorawan_connector_mongodb:init:31 Connecting Dumper to mongodb demonstrator.ieh.uni-stuttgart.de:27017
2018-07-31 17:32:58.871 [warning] <0.7365.1>@lorawan_connector_mongodb:log_termination:81 Connector Dumper terminated: {badarg,[{erlang,apply,[{mongoapi,['Dumper',<<"ghi_temp">>]},createCollection,[]],[]},{lorawan_connector_mongodb,store_fields,3,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_mongodb.erl"},{line,99}]},{lorawan_connector_mongodb,handle_info,2,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_mongodb.erl"},{line,61}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,637}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,711}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,249}]}]}
2018-07-31 17:32:58.882 [debug] <0.7388.1>@lorawan_connector_mongodb:init:31 Connecting Dumper to mongodb demonstrator.ieh.uni-stuttgart.de:27017
2018-07-31 17:32:59.071 [debug] <0.7387.1>@lorawan_mac_commands:build_fopts:74 26011936 <- [dev_status_req]

To me it looks like the connection to the mongoDB is working, but the parsing of the "Publish Uplinks" field is incorrect and does not yield a collection name (only the first part is considered and used as the db name - "**ghi_temp**" in my case)
I am running the latest version (0.5.5). I hope somebody can help me with this...

[image]

[image]

gotthardp, 2018-07-31:
I found another mongodb related error a user was complaining about, but I don't think this will fix your issue. What Erlang version are you using? Could you send me that part of your log which says `Supervisor lorawan_connector_sup started lorawan_connector_mongodb`, please?

Heiner92, 2018-08-01:
I am running erlang version 20.3.6

the debug log you requested (had to restart the server):

> 2018-08-01 11:29:02.116 [debug] <0.178.0>@lager_handler_watcher:119 Lager installed handler {lager_file_backend,"error.log"} into lager_event
2018-08-01 11:29:02.388 [info] <0.328.0> server started
2018-08-01 11:29:02.403 [debug] <0.335.0>@lorawan_http_registry:compile_routes:76 New routes []
2018-08-01 11:29:02.413 [debug] <0.338.0>@lorawan_connector_sup:start_child:16 Start **<<"Django">> (Http Connector - working)**
2018-08-01 11:29:02.413 [debug] <0.340.0>@lorawan_connector_http:connect:157 Connecting Django to https://webhook.site
2018-08-01 11:29:02.415 [debug] <0.338.0>@lorawan_connector_sup:start_child:16 Start {mongodb,**<<"Dumper">>} (MongoDB Connector -  not working)**
2018-08-01 11:29:02.415 [debug] <0.341.0>@lorawan_connector_mongodb:init:31 Connecting Dumper to mongodb demonstrator.ieh.uni-stuttgart.de:27017
2018-08-01 11:29:02.558 [debug] <0.340.0>@lorawan_connector_http:connect:168 Django connected to https://webhook.site (http)
2018-08-01 11:29:02.607 [debug] <0.173.0>@lager_handler_watcher:119 Lager installed handler lager_backend_throttle into lager_event

Thank you very very much! I appreciate all the work you put into this project!

gotthardp, 2018-08-01:
I was expecting one more line, really including `Supervisor lorawan_connector_sup started lorawan_connector_mongodb:start_link` listing all the parameters seen by the module.
Perhaps the line was removed by the throttling mechanism. Could you please uncomment the `error_logger_hwm` line in your sys.config, please, and redo the logs?
https://github.com/gotthardp/lorawan-server/blob/v0.5.5/lorawan_server.config#L34

Heiner92, 2018-08-01:
I downgraded erlang to v20 and everything is working now.
I had one package missing before installing lorawan-server.deb. It installed v21 of the missing package and automatically upgraded all other packages to v21 as well...

gotthardp, 2018-08-01:
I see. Erlang 21 is not supported yet.

closed 2018-08-01 by gotthardp

cocoaway, 2019-09-10:
Hello,

same problem, latest version of the server, compiled with Erlang / OTP 22 [erts-10.4.4]. Running on Ubuntu 19.04.

To which version should I downgrade to ?

Thanks.

altishchenko, 2019-09-10:
Hello, I am using OTP/22 erts-10.4 - no problems with mongo whatsoever... Ubuntu server is **18**.4 though. Erlang is installed from erlang-solutions.

cocoaway, 2019-09-10:
Hello, this is the log . (**myserver-name**  was added by me instead of the real address )

2019-09-10 16:09:27.367 [error] <0.31300.4>@lorawan_application_backend:parse_payload:359 Non ASCII string: <<1,35,69,171,186>>

2019-09-10 16:09:27.569 [warning] <0.31188.4>@lorawan_connector_mongodb:log_termination:94 Connector MongoDB terminated: {function_clause,[{bson,encode_element,[{<<"time">>,{{2019,9,10},{16,9,26.335154}}}],[{file,"/lorawan-server/_build/default/lib/erlmongo/src/bson.erl"},{line,32}]},{bson,'-encode/2-fun-0-',3,[{file,"/lorawan-server/_build/default/lib/erlmongo/src/bson.erl"},{line,17}]},{lists,foldl,3,[{file,"lists.erl"},{line,1263}]},{bson,encode,2,[{file,"/lorawan-server/_build/default/lib/erlmongo/src/bson.erl"},{line,17}]},{bson,encode_element,1,[{file,"/lorawan-server/_build/default/lib/erlmongo/src/bson.erl"},{line,39}]},{bson,'-encode/2-fun-0-',3,[{file,"/lorawan-server/_build/default/lib/erlmongo/src/bson.erl"},{line,17}]},{lists,foldl,3,[{file,"lists.erl"},{line,1263}]},{bson,encode,2,[{file,"/lorawan-server/_build/default/lib/erlmongo/src/bson.erl"},{line,17}]}]}

2019-09-10 16:09:27.570 [error] <0.31188.4>@bson:encode_element:32 gen_server <0.31188.4> terminated with reason: no function clause matching bson:encode_element({<<"time">>,{{2019,9,10},{16,9,26.335154}}}) line 32

2019-09-10 16:09:27.570 [error] <0.31188.4>@bson:encode_element:32 CRASH REPORT Process <0.31188.4> with 0 neighbours crashed with reason: no function clause matching bson:encode_element({<<"time">>,{{2019,9,10},{16,9,26.335154}}}) line 32

2019-09-10 16:09:27.571 [error] <0.400.0>@bson:encode_element:32 Supervisor lorawan_connector_sup had child {mongodb,<<"MongoDB">>} started with lorawan_connector_mongodb:start_link({connector,<<"MongoDB">>,<<"gotoMQTTserver">>,<<"json">>,<<"mongodb://**myserver-name**:27017">>,undefined,...}) at <0.31188.4> exit with reason no function clause matching bson:encode_element({<<"time">>,{{2019,9,10},{16,9,26.335154}}}) line 32 in context child_terminated

2019-09-10 16:09:27.572 [debug] <0.31301.4>@lorawan_connector_mongodb:init:31 Connecting MongoDB to mongodb **myserver-name**:27017

altishchenko, 2019-09-10:
Can you share the real JSON you are trying to store? bson encoder, alas, can't handle lists... and strings must be **binary** strings like <<"string">> in Erlang, not just "string".

altishchenko, 2019-09-10:
Also, your connector screenshot will be helpful.

cocoaway, 2019-09-10:
Hi, screenshot attached .
[image]

cocoaway, 2019-09-11:
Test in handler correctly sends a message to mongodb.
I think that error is in <<"time">>  and not related to the json.

To be sure I reduced the uplink fields, as attached.
[image]

2019-09-11 09:15:23.719 [error] <0.10396.5>@bson:encode_element:32 gen_server <0.10396.5> terminated with reason: no function clause matching bson:encode_element({<<"time">>,{{2019,9,11},{9,15,22.477938}}}) line 32

[error] <0.10396.5>@bson:encode_element:32 CRASH REPORT Process <0.10396.5> with 0 neighbours crashed with reason: no function clause matching bson:encode_element({<<"time">>,{{2019,9,11},{9,15,22.477938}}}) line 32

altishchenko, 2019-09-11:
I can see from the logs that somewhere in the code call for 'time=>iso8601:format(Time)' is missing and this can happen if ASCII payload processing is doing something special... You can try setting 'Custom binary' instead, just to make sure it is the case.
I suggest you open a new issue, given the problem is quite different from the one above!

Petr (@gotthardp)?

cocoaway, 2019-09-11:
Hi, with custom binary 

2019-09-11 15:11:54.208 [warning] <0.19460.5>@lorawan_connector_mongodb:log_termination:94 Connector MongoDB terminated: {badarg,[{bson,encode_cstring,1,[{file,"/lorawan-server/_build/default/lib/erlmongo/src/bson.erl"},{line,184}]},{bson,encode_element,1,[{file,"/lorawan-server/_build/default/lib/erlmongo/src/bson.erl"},{line,52}]},{bson,'-encode/2-fun-0-',3,[{file,"/lorawan-server/_build/default/lib/erlmongo/src/bson.erl"},{line,17}]},{lists,foldl,3,[{file,"lists.erl"},{line,1263}]},{bson,encode,2,[{file,"/lorawan-server/_build/default/lib/erlmongo/src/bson.erl"},{line,17}]},{bson,encode_element,1,[{file,"/lorawan-server/_build/default/lib/erlmongo/src/bson.erl"},{line,39}]},{bson,'-encode/2-fun-0-',3,[{file,"/lorawan-server/_build/default/lib/erlmongo/src/bson.erl"},{line,17}]},{lists,foldl,3,[{file,"lists.erl"},{line,1263}]}]}

2019-09-11 15:11:54.209 [error] <0.19460.5>@bson:encode_cstring:184 gen_server <0.19460.5> terminated with reason: bad argument in bson:encode_cstring/1 line 184

2019-09-11 15:11:54.209 [error] <0.19460.5>@bson:encode_cstring:184 CRASH REPORT Process <0.19460.5> with 0 neighbours crashed with reason: bad argument in bson:encode_cstring/1 line 184

2019-09-11 15:11:54.210 [error] <0.400.0>@bson:encode_cstring:184 Supervisor lorawan_connector_sup had child {mongodb,<<"MongoDB">>} started with lorawan_connector_mongodb:start_link({connector,<<"MongoDB">>,<<"gotoMQTTserver">>,<<"raw">>,<<"mongodb://myserver:27017">>,undefined,...}) at <0.19460.5> exit with reason bad argument in bson:encode_cstring/1 line 184 in context child_terminated

altishchenko, 2019-09-11:
Can you share a resulting JSON, from the MQTT server for example? Also, can you share your Mongo connector screenshot as well?

bachehkaraji, 2019-11-15:
Hi all.
Here is the error log I get

```
lorawan_connector_mongodb:init:31 Connecting mongo to mongodb localhost:27017
...
lorawan_connector_mongodb:store_fields:112 gen_server <0.19076.2> terminated with reason: bad argument in call to erlang:apply({mongoapi,[mongo,<<"test">>]}, createCollection, [<<"tst1">>]) in lorawan_connector_mongodb:store_fields/3 line 112
lorawan_connector_mongodb:store_fields:112 CRASH REPORT Process <0.19076.2> with 0 neighbours crashed with reason: bad argument in call to erlang:apply({mongoapi,[mongo,<<"test">>]}, createCollection, [<<"tst1">>]) in lorawan_connector_mongodb:store_fields/3 line 112
lorawan_connector_mongodb:store_fields:112 Supervisor lorawan_connector_sup had child {mongodb,<<"mongo">>} started with lorawan_connector_mongodb:start_link({connector,<<"mongo">>,<<"Test">>,<<"json">>,<<"mongodb://localhost:27017">>,undefined,<<"test/tst...">>,...}) at <0.19076.2> exit with reason bad argument in call to erlang:apply({mongoapi,[mongo,<<"test">>]}, createCollection, [<<"tst1">>]) in lorawan_connector_mongodb:store_fields/3 line 112 in context child_terminated
...
```

Is there anyone out there, that have a solution for this problem?