gotthardp/lorawan-server #433: RN2483 throwing mac_err instead of max_tx_ok when message successfully transmitted

Gligor721, 2018-07-10:
Hello,
I`m using this server and RN2483 and if the FCnt Up is greater then 16383 I get mac_err instead of max_tx_ok although the message was successfully transmitted.I see the message in server but I get mac_err.
I reset the FCnt Up  to zero and all is working fine i get mac_tx_ok.
Why is this happen?Is problem the server or server configuration?In profile that I use FCnt Check is disabled.
What is the problem?

Gligor721, 2018-07-10:
When my node(rn2483) send message to the server I get mac_err and I see that the message is received by the server(see in the received packet).When I enter a 0 for the counter in the server when i send next message i received mac_tx_ok.What settings i need to change?

Gligor721, 2018-07-10:
I have disabled for the FCnt Check.When I set it reset to zero message is received by the server(view in the received packet) but packet is not received in my app that use websocket. When I return to disabled my app start to received the packet.

gotthardp, 2018-07-10:
You should not disable FCnt Check. What happens if you set Reset on Zero and restart your device?

Gligor721, 2018-07-10:
I set FCnt Check to reset on zero and packets are received by server I see in received packet but packet is not send by web socket to my app, then i return to disabled and packet is send by web socket to my app.I dont reset the device but i will test with reset tomorrow.

Gligor721, 2018-07-16:
If I set FCnt Check to reset on zero I get mac_err but I view the packet to the received packet but packet is not fire via websocket.When I return to disable the FCnt Check packet is received and I get tx_ok.But when the number of packet is greater then 16300 I get mac_err on the node but packet is in the received packet on the server.I think that packet is send by the node and received by the server but ack is not send by the server and node fire mac_err.

Gligor721, 2018-07-16:
Yes I send uplink as "confirmed".
I received all packet via websocket and I see all packet to ""received frames" and all is working fine.I saw that I received mac_err on node side when the number of packet send (FCnt Check) is greater then 16300.
Then I change the setting in the profile from disable to reset to zero and with that setting I dont get any packet via web socket and get mac_err but in the received frames i get the packets.Than i return the setting and all is working.
All is working fine uplink and downlink until the counter is greater then 16300.(approximately)
I power the lora node send the information and disable power to the node and this is repeated,Because of that the counter is always is zero. Why server not send ack after counter get greater then xxx?

gotthardp, 2018-07-16:
How could you get counter to 16300 when you always power on and off the device?

Gligor721, 2018-07-16:
The counter is from the server not in the node.This number I see in the server.

Gligor721, 2018-07-16:
In the server counter is growing after packet is received from the node.

Gligor721, 2018-07-16:
If I enable the FCnt Check I will not get the packet because I send packet always with zero counter and the server is check this number if FCnt Check is enabled.

gotthardp, 2018-07-16:
Alright, I understand now. Do you really have to always reset the counter? Is it because of a bug, is it not yet implemented, or from some reason you have to always reset? The LoRaWAN standard says the FCnt shall be persistent.

Gligor721, 2018-07-16:
I make sensor that work on battery and work in following way:
sensor is read (magnetic field) and calculate  and send the information to the server (my app) and go to sleep but before sleep i turn off all the peripheral device so I can save the power and then controller go to sleep.After 20 second timer is fire and controller is wake up from sleep and than read calculate and if is change send to the server.Only when i need to send I power up the lora module initialize send and power off.

gotthardp, 2018-07-16:
It's not a standard-compliant behavior, but I made a "enhancement" for you. Sending Uplink FCnt=0 with FCntCheck=disabled will reset the FCnt Down counter too, so it shall work for you as you expect.

The fix is made in the master branch, so you will need to build it yourself and then add then add the Group/Region configuration introduced in release 0.6.

Gligor721, 2018-07-16:
I am glad the problem is solved. Thank you for all the support.  I will test the solution and let you know.
When can I except the version 0.6 to be officially published ?

gotthardp, 2018-07-16:
I'd say around September-October

closed 2018-08-01 by gotthardp