gotthardp/lorawan-server #434: Tx_power error

gmbankar37, 2018-07-12:
I am using the network IN865. I have set tx_power=20 and I have already joined on a node on a gateway (say gateway A). When I am trying to connect it with other gateway with same parameters it is giving me error of tx_power and also all all downlinks are lost. This results in repeated join requests but the node is not able to join. What might be the problem?

gotthardp, 2018-07-12:
This is a gateway issue.
X_POWER = Rejected because requested power is not supported by gateway

gmbankar37, 2018-07-12:
Even if I am setting the tx_power to 0 it is giving me same error.

gotthardp, 2018-07-12:
What about values > 0?
Is there some error indication in the gateway log?

gmbankar37, 2018-07-12:
Yes same with values greater than 0. yes just indicates error tx_power. It
is also visible on the time line.

gotthardp, 2018-07-12:
I mean-- is there something in the packet_forwarder log? The timeline is shown by the server; since this is a gateway error you should check for gateway logs.

gmbankar37, 2018-07-13:
This is the error log of the work I did yesterday.

2018-07-11 10:09:19.581 [error] <0.281.0>@lorawan_connector_mqtt:handle_reconnect:258 gen_server <0.281.0> terminated with reason: bad record connector in lorawan_connector_mqtt:handle_reconnect/3 line 258
2018-07-11 10:09:19.609 [error] <0.281.0>@lorawan_connector_mqtt:handle_reconnect:258 CRASH REPORT Process <0.281.0> with 0 neighbours exited with reason: bad record connector in lorawan_connector_mqtt:handle_reconnect/3 line 258 in gen_server:terminate/7 line 812
2018-07-11 10:09:19.613 [error] <0.278.0>@lorawan_connector_mqtt:handle_reconnect:258 Supervisor lorawan_connector_sup had child {mqtt,<<"MQTT_PI">>} started with lorawan_connector_mqtt:start_link({connector,<<"MQTT_PI">>,<<"TestNew">>,<<"json">>,<<"mqtt://127.0.0.1:1883">>,<<"out/{devaddr}">>,...}) at <0.281.0> exit with reason bad record connector in lorawan_connector_mqtt:handle_reconnect/3 line 258 in context child_terminated
2018-07-11 12:10:35.868 [error] <0.285.0>@lorawan_connector_mqtt:handle_reconnect:258 gen_server <0.285.0> terminated with reason: bad record connector in lorawan_connector_mqtt:handle_reconnect/3 line 258
2018-07-11 12:10:35.892 [error] <0.285.0>@lorawan_connector_mqtt:handle_reconnect:258 CRASH REPORT Process <0.285.0> with 0 neighbours exited with reason: bad record connector in lorawan_connector_mqtt:handle_reconnect/3 line 258 in gen_server:terminate/7 line 812
2018-07-11 12:10:35.894 [error] <0.282.0>@lorawan_connector_mqtt:handle_reconnect:258 Supervisor lorawan_connector_sup had child {mqtt,<<"MQTT_PI">>} started with lorawan_connector_mqtt:start_link({connector,<<"MQTT_PI">>,<<"TestNew">>,<<"json">>,<<"mqtt://127.0.0.1:1883">>,<<"out/{devaddr}">>,...}) at <0.285.0> exit with reason bad record connector in lorawan_connector_mqtt:handle_reconnect/3 line 258 in context child_terminated
2018-07-11 13:08:49.569 [error] <0.574.0>@lorawan_connector_mqtt:handle_reconnect:258 gen_server <0.574.0> terminated with reason: bad record connector in lorawan_connector_mqtt:handle_reconnect/3 line 258
2018-07-11 13:08:49.572 [error] <0.574.0>@lorawan_connector_mqtt:handle_reconnect:258 CRASH REPORT Process <0.574.0> with 0 neighbours exited with reason: bad record connector in lorawan_connector_mqtt:handle_reconnect/3 line 258 in gen_server:terminate/7 line 812
2018-07-11 13:08:49.574 [error] <0.282.0>@lorawan_connector_mqtt:handle_reconnect:258 Supervisor lorawan_connector_sup had child {mqtt,<<"MQTT_PI">>} started with lorawan_connector_mqtt:start_link({connector,<<"MQTT_PI">>,<<"TestNew">>,<<"json">>,<<"mqtt://127.0.0.1:1883">>,<<"out/{devaddr}">>,...}) at <0.574.0> exit with reason bad record connector in lorawan_connector_mqtt:handle_reconnect/3 line 258 in context child_terminated
2018-07-11 15:35:53.115 [error] <0.285.0>@lorawan_connector_mqtt:handle_reconnect:258 gen_server <0.285.0> terminated with reason: bad record connector in lorawan_connector_mqtt:handle_reconnect/3 line 258
2018-07-11 15:35:53.138 [error] <0.285.0>@lorawan_connector_mqtt:handle_reconnect:258 CRASH REPORT Process <0.285.0> with 0 neighbours exited with reason: bad record connector in lorawan_connector_mqtt:handle_reconnect/3 line 258 in gen_server:terminate/7 line 812
2018-07-11 15:35:53.140 [error] <0.282.0>@lorawan_connector_mqtt:handle_reconnect:258 Supervisor lorawan_connector_sup had child {mqtt,<<"MQTT_PI">>} started with lorawan_connector_mqtt:start_link({connector,<<"MQTT_PI">>,<<"TestNew">>,<<"json">>,<<"mqtt://127.0.0.1:1883">>,<<"out/{devaddr}">>,...}) at <0.285.0> exit with reason bad record connector in lorawan_connector_mqtt:handle_reconnect/3 line 258 in context child_terminated
2018-07-11 16:19:07.097 [error] <0.285.0>@lorawan_connector_mqtt:handle_reconnect:258 gen_server <0.285.0> terminated with reason: bad record connector in lorawan_connector_mqtt:handle_reconnect/3 line 258
2018-07-11 16:19:07.118 [error] <0.285.0>@lorawan_connector_mqtt:handle_reconnect:258 CRASH REPORT Process <0.285.0> with 0 neighbours exited with reason: bad record connector in lorawan_connector_mqtt:handle_reconnect/3 line 258 in gen_server:terminate/7 line 812
2018-07-11 16:19:07.122 [error] <0.282.0>@lorawan_connector_mqtt:handle_reconnect:258 Supervisor lorawan_connector_sup had child {mqtt,<<"MQTT_PI">>} started with lorawan_connector_mqtt:start_link({connector,<<"MQTT_PI">>,<<"TestNew">>,<<"json">>,<<"mqtt://127.0.0.1:1883">>,<<"out/{devaddr}">>,...}) at <0.285.0> exit with reason bad record connector in lorawan_connector_mqtt:handle_reconnect/3 line 258 in context child_terminated
2018-07-11 17:09:33.072 [error] <0.277.0> gateway B827EBFFFEF7A67C unknown_mac
2018-07-11 17:09:43.148 [error] <0.277.0> gateway B827EBFFFEF7A67C unknown_mac
2018-07-11 17:09:53.228 [error] <0.277.0> gateway B827EBFFFEF7A67C unknown_mac
2018-07-11 17:10:03.132 [error] <0.277.0> gateway B827EBFFFEF7A67C unknown_mac
2018-07-11 17:10:03.318 [error] <0.277.0> gateway B827EBFFFEF7A67C unknown_mac
2018-07-11 17:10:13.398 [error] <0.277.0> gateway B827EBFFFEF7A67C unknown_mac
2018-07-11 17:10:23.478 [error] <0.277.0> gateway B827EBFFFEF7A67C unknown_mac
2018-07-11 17:10:33.075 [error] <0.277.0> gateway B827EBFFFEF7A67C unknown_mac
2018-07-11 17:10:33.568 [error] <0.277.0> gateway B827EBFFFEF7A67C unknown_mac
2018-07-11 17:10:43.648 [error] <0.277.0> gateway B827EBFFFEF7A67C unknown_mac
2018-07-11 17:10:53.728 [error] <0.277.0> gateway B827EBFFFEF7A67C unknown_mac
[... cut ...]
2018-07-12 12:58:35.155 [error] <0.293.0> node 028C3FA3 <<"tx_power">>
2018-07-12 12:58:42.211 [error] <0.293.0> node 028C3FA3 <<"tx_power">>
2018-07-12 13:23:34.012 [error] <0.291.0> node 028C3FA3 <<"tx_power">>
2018-07-12 13:23:41.001 [error] <0.291.0> node 028C3FA3 <<"tx_power">>
2018-07-12 13:23:48.045 [error] <0.291.0> node 028C3FA3 <<"tx_power">>
2018-07-12 13:23:55.093 [error] <0.291.0> node 028C3FA3 <<"tx_power">>
2018-07-12 13:24:02.148 [error] <0.291.0> node 028C3FA3 <<"tx_power">>
2018-07-12 13:24:09.196 [error] <0.291.0> node 028C3FA3 <<"tx_power">>

gmbankar37, 2018-07-13:
This is what my dashboard shows when I send a join command.

gotthardp, 2018-07-13:
But that's the SERVER log. You need a GATEWAY log-- the log that is stored on the LoRaWAN Gateway, the one generated by semtech packet_forwarder.

gmbankar37, 2018-07-13:
where will I find those logs?

gotthardp, 2018-07-13:
What gateway do you have? Try looking somewhere in /var/log..... on the gateway

gmbankar37, 2018-07-13:
I am using RAK831. I havent found anything in /var/log. I have two gateways
(two RAK831) and it connects to one but gives the error to other. Can a
single node connect to two gateways on two different server (not at the
same time)?

gotthardp, 2018-07-13:
You must have only one server.

gotthardp, 2018-08-02:
Did you manage to solve this issue? Shall I close it?

gmbankar37, 2018-08-02:
Yes you can close it.

closed 2018-08-02 by gotthardp

keyurhc, 2018-12-28:
I am also facing the same issue of **tx_power** error in events log and I am using **Friendcom** gateway. I have configured my frequency as _**Indian**_ frequency.Please reply as soon as possible @gotthardp.

gotthardp, 2018-12-28:
Hello. The error message is generated by the gateway, so you will first need to figure out why is the gateway rejecting the tx-power used.

keyurhc, 2018-12-28:
How can I check that.Any ways plz suggest me@gotthatdp

keyurhc, 2018-12-28:
Sorry it’s @gotthardp

gotthardp, 2018-12-28:
Start by reviewing the gateway logs or by contacting the gateway manufacturer support team.

gmbankar37, 2018-12-31:
> Ohk thanks a lot@gotthardp

Go check the global_config file. Check the LUT tx_power. If the
transmitting power of gateway doesn't match with the power in LUT table
then it gives this error or rejects it. Adjust the gateway power and you
wont get that error again.