gotthardp/lorawan-server #436: Enhancement: Add "App Argument" column to Received Frames and Filter by DevEUI or App Argument

cmenscher, 2018-07-18:
It would be nice to see App Arguments for particular devices in the Received Frames. We incorporate installed location data in the arguments, for example, so it would help us identify frames for particular customer locations.  We would also love to see device DevEUI's as a column.

Being able to filter on these would be fantastic, as well.

gotthardp, 2018-07-18:
I did two thirds: The new release (0.6) will also log the "App Arguments" and also filtering based on that will be possible. Logging DevEUI is not that easy, so I defer that for later.

cmenscher, 2018-07-18:
I'll take it!  Thanks. When is 0.6 scheduled for release?

gotthardp, 2018-07-19:
Probably in the Sep-Oct timeframe.

closed 2018-07-19 by gotthardp