gotthardp/lorawan-server #437: Updating App Arguments field in Commissioned devices is not reflected in Activated Nodes, and vice versa

cmenscher, 2018-07-20:
We use App Arguments to record installed location data for each device.  This is often configured AFTER we have commissioned a device in lorawan-server.  When the App Argument field is updated after a device has joined (via OTAA in our case), we are forced to update the App Argument in both the Commissioned and Activated Nodes detail pages.  (We update in Commissioned so that in the event of subsequent device resets or reprogramming, the correct app argument will be used on any new joins.)

The expected behavior is that updating app arguments on either page will update the app arguments in both.  This may be intentional behavior, but we are having a hard time understanding why it may be the case.  The only thing we can think of is that it may be desirable to have a different set of app arguments upon commissioning as an initial state.

gotthardp, 2018-07-20:
Hello. I understand your need, but there is a certain logic behind the current mechanism: the server distinguishes _effective_ parameters (those in Activated Nodes) and "to be" _desired_ parameters that will become effective with the next reboot or join.
Would creating a Node attribute `location` solve your problem? If the parameter survives rejoin/reboot, of course.

cmenscher, 2018-07-20:
A location field would definitely be useful, but we're not using GPS coords...we're using street addresses.  So a flexible text field is more useful to us.  We're really doing this so we can sort the Commissioned and Activated lists by location to view them all as a group...we're not really using it in the application backend.  It would also help to be able to sort by this App Argument/location on the Received Frames page... 😎

gotthardp, 2018-07-20:
I had in mind a simple text field `location` but for an Activated Node only. Commissioned Devices would have no location (to avoid inconsistency ;-).
The Received Frames field is not infinite. I cannot keep adding fields, but I can remove App Arg and put Location instead. It may actually make even more sense.

cmenscher, 2018-07-20:
Well the nice thing about the app argument field is that it doesn't have a specific context.  You can use it for whatever you want...location is constrained that way.  For example, when we were testing some new devices on our network we used the app argument to quickly identify which device type was reporting without having to look up a DevAddr or DevEUI.  Handy.  But like I mentioned, our primary goal is to be able to view devices and traffic associated with a specific installed address, so we're open to any way to do that.  We've even considered using separate profiles, but I don't think that's the appropriate use.

hallard, 2019-07-31:
Guys,
I may have misunderstood something, the `location` field is not pushed on uplink? I can't find it in the handler.

gotthardp, 2019-07-31:
It is not.