gotthardp/lorawan-server #439: Not receiving uplink frames

gmbankar37, 2018-07-24:
I upgraded from 0.5.3 to 0.5.5. I copied the mnesia database folder to have same configuration as before. I am able to join the node to the gateway. When I am sending a frame the gateway is not able to receive it. It shows nothing in received frame but i get an acknowledgment in the terminal that the frame is received. Is the gateway to server connection faulty? but I am getting a join on the timeline.

gotthardp, 2018-07-24:
what kind of "acknowledgment in the terminal that the frame is received" do you mean?

gmbankar37, 2018-07-24:
Yes.

gmbankar37, 2018-07-24:
when I send at+send=1,2,"data" I get at+recv=1,0,0

gotthardp, 2018-07-24:
What application are you using?

gmbankar37, 2018-07-24:
> On Tue, Jul 24, 2018 at 2:32 PM, gaurav bankar <gmbankar37@gmail.com>

I created a new one in handler. Passing the uplink frame as it is. also
have a mqtt connector.

gmbankar37, 2018-07-24:
Got it working. Dont know the logic behind it but first I tried to send
with semtech-mote application and then I changed it to my handler
application.

gotthardp, 2018-07-24:
Problem solved, then :)

closed 2018-07-24 by gotthardp