gotthardp/lorawan-server #440: Increase pagination size

cmenscher, 2018-07-24:
Most of the tables in the application paginate their lists at 30 rows.  We would love to be able to adjust that to be much higher...100, 200, or even unlimited/unpaginated.

gotthardp, 2018-07-24:
Why?

cmenscher, 2018-07-24:
Well, for example, the Received Frames history doesn't go back particularly far per page, and we are just getting started installing devices.  So trying to track down a set of payloads can require a bunch of clicking.  And the device list, even with filters, has many pages to click through.  Since it's just text I'd rather be able to scroll quickly through a large list, or use a simple find on page, to get to a particular item (i.e. find a devEUI or devAddr) or set of items.

We can export the lists into CSV's and open as a big spreadsheet, but that's not ideal, especially if we're looking to navigate into a device detail page in the application.

gotthardp, 2018-07-24:
Alright. I will try making some configuration parameter for that.

gotthardp, 2018-07-26:
I just implemented that. It will be in 0.6.0.

closed 2018-07-26 by gotthardp