gotthardp/lorawan-server #444: Compatibility with TTN gateway connector protocol?

sealj553, 2018-07-27:
Are there any plans to add compatibility with TheThingsNetwork's [Gateway Connector Protocol](https://www.thethingsnetwork.org/docs/gateways/start/connection.html) (protocol buffers over RPC or MQTT)? I'm working on a gateway targeted for TTN, but It would be nice if this platform would be compatible as well. I'd like to help, but I don't know any Erlang.

gotthardp, 2018-07-27:
Hello. This server supports the "plain old" packet_forwarder protocol. Your gateway will not support that?

sealj553, 2018-07-27:
It does at the moment, but it looks like TTN is moving away from that protocol. They're working on TTN v3, and it's probable that they might drop support for the old protocol. For my gateway, I'd like to use the newer protocol in the name of future compatibility, performance, and security.

The reasons not to use the old protocol mentioned on TTN's site are:
> It does not provide authentication. Gateways are supposed to identify themselves with their EUI - but anyone can usurpate another gateway.
> There is no encryption available. Gateway messages can be intercepted - and modified - during transport.
> Message exchange is not reliable, since the exchange is over 2-way UDP.
> The UDP protocol is based on a format close to JSON, but not fully compliant ; for example, the specs specify that an object can hold multiple properties with the same index. This makes it easy to write by hand for testing purposes, but difficult to parse in some situations.

gotthardp, 2018-07-27:
I see. Well, this server is intended for private networks, so the gateway to server communication is assumed to be within a VPN, so many of the drawbacks mentioned are not so critical.
I will probably wait a bit to see how many users request it, hoping that the LoRaWAN alliance will develop some standard for that.

gotthardp, 2019-05-23:
Please note that a new protocol and a new gateway firmware was implemented, which is supported by TTN and (since v0.7.0) also by this server: https://doc.sm.tc/station/tcproto.html
The protocol fixes the deficiencies listed above (TCP-base, with SSL and authentication).

closed 2019-05-23 by gotthardp