gotthardp/lorawan-server #445: Connectivity issue

alireza2980, 2018-07-28:
I want to verify if frames are being sent to a server , I used https://requestbin.fullcontact.com/ in connector and left all other fields empty 
[image]

I setup the handler as follow too 
[image]

but yet I receive nothing on the listening bin.
gateway receives frames 
[image]

and no connectivity issues (using curl messages are being sent) , my error log is as follow ( which does not seem to be relevant to the issue) 

2018-07-28 18:24:44.355 [debug] <0.273.0> Supervisor mongodb_supervisor started mongodb:start() at pid <0.274.0>
2018-07-28 18:24:44.360 [info] <0.33.0> Application erlmongo started on node 'lorawan@1gate-b827eb7ff5ea'
2018-07-28 18:24:44.401 [debug] <0.293.0> Supervisor {<0.293.0>,ranch_listener_sup} started ranch_conns_sup:start_link(http, supervisor, 5000, ranch_tcp, 5000, cowboy_clear) at pid <0.294.0>
2018-07-28 18:24:44.408 [debug] <0.295.0> Supervisor {<0.295.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5593>, ranch_tcp, <0.294.0>) at pid <0.296.0>
2018-07-28 18:24:44.408 [debug] <0.295.0> Supervisor {<0.295.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5593>, ranch_tcp, <0.294.0>) at pid <0.297.0>
2018-07-28 18:24:44.408 [debug] <0.295.0> Supervisor {<0.295.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5593>, ranch_tcp, <0.294.0>) at pid <0.298.0>
2018-07-28 18:24:44.409 [debug] <0.295.0> Supervisor {<0.295.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5593>, ranch_tcp, <0.294.0>) at pid <0.299.0>
2018-07-28 18:24:44.409 [debug] <0.295.0> Supervisor {<0.295.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5593>, ranch_tcp, <0.294.0>) at pid <0.300.0>
2018-07-28 18:24:44.409 [debug] <0.295.0> Supervisor {<0.295.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5593>, ranch_tcp, <0.294.0>) at pid <0.301.0>
2018-07-28 18:24:44.410 [debug] <0.295.0> Supervisor {<0.295.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5593>, ranch_tcp, <0.294.0>) at pid <0.302.0>
2018-07-28 18:24:44.410 [debug] <0.295.0> Supervisor {<0.295.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5593>, ranch_tcp, <0.294.0>) at pid <0.303.0>
2018-07-28 18:24:44.410 [debug] <0.295.0> Supervisor {<0.295.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5593>, ranch_tcp, <0.294.0>) at pid <0.304.0>
2018-07-28 18:24:44.411 [debug] <0.295.0> Supervisor {<0.295.0>,ranch_acceptors_sup} started ranch_acceptor:start_link(#Port<0.5593>, ranch_tcp, <0.294.0>) at pid <0.305.0>
2018-07-28 18:24:44.411 [debug] <0.293.0> Supervisor {<0.293.0>,ranch_listener_sup} started ranch_acceptors_sup:start_link(http, 10, ranch_tcp, [{connection_type,supervisor},{port,8080}]) at pid <0.295.0>
2018-07-28 18:24:44.412 [debug] <0.249.0> Supervisor ranch_sup started ranch_listener_sup:start_link(http, 10, ranch_tcp, [{connection_type,supervisor},{port,8080}], cowboy_clear, #{connection_type => supervisor,env => #{dispatch => [{'_',[],[{[<<"api">>,<<"servers">>],[],...},...]}]},...}) at pid <0.293.0>
2018-07-28 18:24:44.412 [info] <0.306.0> server started
2018-07-28 18:24:44.441 [debug] <0.306.0> Supervisor {<0.306.0>,lorawan_sup} started lorawan_db_guard:start_link() at pid <0.308.0>
2018-07-28 18:24:44.450 [debug] <0.309.0> Supervisor {<0.309.0>,lorawan_gw_sup} started lorawan_gw_router:start_link() at pid <0.310.0>
2018-07-28 18:24:44.461 [debug] <0.309.0> Supervisor {<0.309.0>,lorawan_gw_sup} started lorawan_gw_forwarder:start_link([{port,1680}]) at pid <0.311.0>
2018-07-28 18:24:44.464 [debug] <0.309.0> Supervisor {<0.309.0>,lorawan_gw_sup} started lorawan_handler_sup:start_link() at pid <0.312.0>
2018-07-28 18:24:44.465 [debug] <0.306.0> Supervisor {<0.306.0>,lorawan_sup} started lorawan_gw_sup:start_link() at pid <0.309.0>
2018-07-28 18:24:44.466 [debug] <0.306.0> Supervisor {<0.306.0>,lorawan_sup} started lorawan_http_registry:start_link() at pid <0.313.0>
2018-07-28 18:24:44.475 [debug] <0.313.0>@lorawan_http_registry:compile_routes:76 New routes []
2018-07-28 18:24:44.480 [debug] <0.314.0> Supervisor {<0.314.0>,lorawan_backend_sup} started lorawan_connector_sup:start_link() at pid <0.315.0>
2018-07-28 18:24:44.485 [debug] <0.314.0> Supervisor {<0.314.0>,lorawan_backend_sup} started lorawan_backend_factory:start_link() at pid <0.316.0>
2018-07-28 18:24:44.486 [debug] <0.306.0> Supervisor {<0.306.0>,lorawan_sup} started lorawan_backend_sup:start_link() at pid <0.314.0>
2018-07-28 18:24:44.486 [info] <0.33.0> Application lorawan_server started on node 'lorawan@1gate-b827eb7ff5ea'
2018-07-28 18:24:54.461 [info] <0.310.0> gateway B827EBFFFE7FF5EA {connected,{{127,0,0,1},41598}}
2018-07-28 18:25:16.554 [error] <0.327.0>@lorawan_mac:ingest_data_frame:92 <<"63EB12D0">> Unexpected mtype 3
2018-07-28 18:25:36.997 [debug] <0.333.0>@lorawan_handler:retransmit:217 EB020016 retransmitting
2018-07-28 18:25:37.001 [error] <0.310.0> node EB020016 <<"tx_power">>
2018-07-28 18:25:39.851 [warning] <0.310.0> gateway B827EBFFFE7FF5EA {downlinks_lost,1}
2018-07-28 18:25:42.041 [debug] <0.335.0>@lorawan_handler:retransmit:217 EB020016 retransmitting
2018-07-28 18:25:42.044 [error] <0.310.0> node EB020016 <<"tx_power">>
2018-07-28 18:25:47.483 [debug] <0.338.0>@lorawan_handler:retransmit:217 EB020016 retransmitting
2018-07-28 18:25:47.486 [error] <0.310.0> node EB020016 <<"tx_power">>
2018-07-28 18:26:09.853 [warning] <0.310.0> gateway B827EBFFFE7FF5EA {downlinks_lost,2}
2018-07-28 18:29:07.980 [error] <0.447.0>@lorawan_mac:ingest_data_frame:92 <<"EA49F0C2">> Unexpected mtype 3
2018-07-28 18:30:26.574 [debug] <0.466.0>@lorawan_handler:retransmit:217 EB020017 retransmitting
2018-07-28 18:30:26.578 [error] <0.310.0> node EB020017 <<"tx_power">>
2018-07-28 18:30:31.651 [debug] <0.468.0>@lorawan_handler:retransmit:217 EB020017 retransmitting
2018-07-28 18:30:31.655 [error] <0.310.0> node EB020017 <<"tx_power">>
2018-07-28 18:30:38.110 [debug] <0.471.0>@lorawan_handler:retransmit:217 EB020017 retransmitting
2018-07-28 18:30:38.113 [error] <0.310.0> node EB020017 <<"tx_power">>
2018-07-28 18:30:39.855 [warning] <0.310.0> gateway B827EBFFFE7FF5EA {downlinks_lost,3}
2018-07-28 18:30:47.292 [debug] <0.59.0> Supervisor kernel_safe_sup started pg2:start_link() at pid <0.495.0>
2018-07-28 18:32:46.607 [debug] <0.509.0>@lorawan_mac:ingest_frame0:59 Unknown frame: <<221,156,79,77,72,86,97,104,13,114,183,123,239,241,58,227,3,14,29,120,130,191,86,92,48,202,233,146,164,111,83,179,227,15,185,30,176,218,152,12,239,25,153,121,187,73,143,232,159,191,116,125,75,78,159,30,237,120,113,187,122,60,28,167,101,49,56,155,43,237,251,134,63,179,21,98,41,85,172,198,145,11,39,73,218,226,244,76,138>>
2018-07-28 18:32:46.607 [error] <0.509.0> server unknown_frame
2018-07-28 18:33:40.741 [error] <0.511.0>@lorawan_mac:ingest_frame:22 gen_statem <0.511.0> in state idle terminated with reason: no match of right hand value <<"آµ51">> in lorawan_mac:ingest_frame/1 line 22
2018-07-28 18:33:41.884 [error] <0.511.0>@lorawan_mac:ingest_frame:22 CRASH REPORT Process <0.511.0> with 0 neighbours crashed with reason: no match of right hand value <<"آµ51">> in lorawan_mac:ingest_frame/1 line 22
2018-07-28 18:33:41.885 [error] <0.312.0>@lorawan_mac:ingest_frame:22 Supervisor lorawan_handler_sup had child handler started with lorawan_handler:start_link() at <0.511.0> exit with reason no match of right hand value <<"آµ51">> in lorawan_mac:ingest_frame/1 line 22 in context child_terminated
2018-07-28 18:35:04.547 [debug] <0.529.0>@lorawan_mac:ingest_frame0:59 Unknown frame: <<46,232,199,104,235,80,249,132,16,80,159,149,119,246,128,225,37,50,40,61,66,84,103,201,132,45,160,10,136,49,108,20,222,125,152,184,99,154,220,79,212,152,170,78,133,251,15,23,66,213,64,29,57,85,49,145,176,253,254,11,174,201,235,93,51,176,54,143,207,41,171,93,92,117,187,149,113,48,23,84,50,3,50,76,218,204,134,83,179,180,75,228,4,55,59,250,19,58,199,47,167,47,43,0,206,184,60,88,147,19,48,97,48,175,126,242,216,47,51,68,120,86,147,37,30,20,9,10,17,105,64,135,76,188,37,84,138,95,27,184,105,88,76,47,63,52,84,205,98,56,241,137,68,79,150,190,193,66,25,67,237,184,93,12,68,58,84,9,13,195,236,193,255,35,137,188,195,8,142,86,154,250,63,26,64,107,131,223,234,248,169,22,114,19,250,187,20,113,29,160,50,122,95,49,72,135,81,187,89,76,115,49,181,179,245,209,154,33,89,26,114,157,80,157,155,116,145,121,216,22,52>>
2018-07-28 18:35:04.547 [error] <0.529.0> server unknown_frame
[... cut ...]
2018-07-28 18:56:22.145 [debug] <0.730.0>@lorawan_handler:retransmit:217 EB020017 retransmitting
2018-07-28 18:56:22.148 [error] <0.310.0> node EB020017 <<"tx_power">>
2018-07-28 18:56:26.598 [debug] <0.731.0>@lorawan_handler:retransmit:217 EB020017 retransmitting
2018-07-28 18:56:26.601 [error] <0.310.0> node EB020017 <<"tx_power">>
2018-07-28 18:56:39.881 [warning] <0.310.0> gateway B827EBFFFE7FF5EA {downlinks_lost,3}
2018-07-28 18:59:31.065 [error] <0.762.0>@lorawan_mac:ingest_data_frame:92 <<"2175455B">> Unexpected mtype 3

The send button is grey offline too.

gotthardp, 2018-07-28:
There are Failed flags raised. The connector isnt active. Also, server docs say there must be no path appended to URI and that you have to set Publish Uplinks path. Fix the config and clear the flags.

gotthardp, 2018-08-02:
Were you able to solve this issue?

gotthardp, 2018-08-10:
I assume the issue is solved.

alireza2980, 2018-08-11:
no actually , yet trying , the server is a simple http server , no setting for down-link , up-link ,etc.

alireza2980, 2018-08-11:
Is it possible to receive payload on sites like request.bin or webhook.site through lorawan-server ?

gotthardp, 2018-08-11:
Yes, that is possible.

closed 2018-08-30 by gotthardp