gotthardp/lorawan-server #447: Run without mnesia

gotthardp, 2018-08-01:
To support clustering functions

altishchenko, 2018-08-02:
Petr, can you clarify for me what are you trying to do?

gotthardp, 2018-08-02:
Right now the server cannot run when mnesia is not running or tables are not loaded. This makes cluster operations harder (sometimes impossible). I want to make it more robust against mnesia related troubles, which appear to be not unusual when you have a cluster.

altishchenko, 2018-08-02:
Shall it not be  better to `wait_for_tables/2`, which is the recommended mnesia startup function? And, as another idea, make disc_copies an option for slave nodes, like RabbitMQ does? Troubles are few and can be comprehended and tackled. Some times just a clear description/documentation will do the trick. The old 0.4 cluster, I was telling you about sometime ago, still works without any issues.

gotthardp, 2018-08-02:
Yeah, that could be the solution. I didn't mean to destroy everything, just to polish and enhance.

altishchenko, 2018-08-02:
Actually, `wait_for_tables/2` is cited as a must in "Erlang and OTP in Action" book by Manning publishing and I use in my other local projects.

gotthardp, 2018-08-02:
Well, I use `wait_for_tables/2`, but perhaps in a wrong way.

BTW, are you making some changes to the clustering, or am I free to modify it? If there is some PR on its way, I can wait.

altishchenko, 2018-08-02:
In my opinion, server startup should fail if `wait_for_tables` fails at least for some tables without which the server cannot operate. 
I am currently studying master/slave mnesia configuration and node addition/removal, also db/node upgrade is an open issue. I'll write you details in an e-mail.
What changes would you like to implement?

gotthardp, 2018-08-02:
Alright, I will wait then. I wanted e.g. do a proper node removal (with mnesia shutdown etc.) But it will be easier with your guidance :)

gotthardp, 2018-08-10:
This is no longer needed. The recent clustering updates solve this issue.

closed 2018-08-10 by gotthardp