gotthardp/lorawan-server #448: Server Admin Web ui

gyha, 2018-08-03:
I started lorawan-server and tried to use web ui.
So i entered the http://192.168.0.62:8080
But nothing appeared in the browser like below picture.
Could you tell me the reason why and how to fix?
[image]

gotthardp, 2018-08-03:
Could you have a look into a developer console of the browser and let me know what is there? Have you tried other browsers?

gyha, 2018-08-03:
I just tried it on the linux (192.168.0.62:8080). It works. But when i try it on the Window, it dosen't work. 
Could you explain about this?

gotthardp, 2018-08-03:
And could you provide the browser console log, please?

gyha, 2018-08-03:
Of course. Here it is.

SCRIPT1014: 유효하지 않은 문자입니다.
파일: admin.js, 줄: 923, 열: 12

SCRIPT5022: [$injector:modulerr] http://errors.angularjs.org/1.6.4/$injector/modulerr?p0=myApp&p1=Error%3A%20%5B%24injector%3Anomod%5D%20http%3A%2F%2Ferrors.angularjs.org%2F1.6.4%2F%24injector%2Fnomod%3Fp0%3DmyApp%0A%20%20%20at%20Anonymous%20function%20(http%3A%2F%2F192.168.0.62%3A8080%2Fadmin%2Fng-admin.min.js%3A6%3A18572)%0A%20%20%20at%20n%20(http%3A%2F%2F192.168.0.62%3A8080%2Fadmin%2Fng-admin.min.js%3A6%3A18081)%0A%20%20%20at%20Anonymous%20function%20(http%3A%2F%2F192.168.0.62%3A8080%2Fadmin%2Fng-admin.min.js%3A6%3A18309)%0A%20%20%20at%20Anonymous%20function%20(http%3A%2F%2F192.168.0.62%3A8080%2Fadmin%2Fng-admin.min.js%3A6%3A26592)%0A%20%20%20at%20a%20(http%3A%2F%2F192.168.0.62%3A8080%2Fadmin%2Fng-admin.min.js%3A6%3A9188)%0A%20%20%20at%20l%20(http%3A%2F%2F192.168.0.62%3A8080%2Fadmin%2Fng-admin.min.js%3A6%3A26435)%0A%20%20%20at%20Be%20(http%3A%2F%2F192.168.0.62%3A8080%2Fadmin%2Fng-admin.min.js%3A6%3A28569)%0A%20%20%20at%20i%20(http%3A%2F%2F192.168.0.62%3A8080%2Fadmin%2Fng-admin.min.js%3A6%3A15972)%0A%20%20%20at%20J%20(http%3A%2F%2F192.168.0.62%3A8080%2Fadmin%2Fng-admin.min.js%3A6%3A16359)%0A%20%20%20at%20Z%20(http%3A%2F%2F192.168.0.62%3A8080%2Fadmin%2Fng-admin.min.js%3A6%3A15618)
파일: ng-admin.min.js, 줄: 6, 열: 26775

gotthardp, 2018-08-03:
My google translator says:
```
SCRIPT1014: Invalid character.
File: admin.js, line: 923, Column: 12
```

What server version are you running and what is on line 923, column 12 in the file `priv/admin/admin.js`, please?

gotthardp, 2018-08-10:
@gyha, any update?