gotthardp/lorawan-server #450: "Data has changed"

danielknox, 2018-08-05:
Possible bug, but whenever I try to update a gateway's metadata (e.g. description / location on map), I regularly get an error stating that the "data has changed. Please reload and repeat the action". I guess this is because the gateway has sent an uplink / status update in the meantime which causes this. Any way of separating this out as it can be a little frustrating updating a bunch of gateways.

gotthardp, 2018-08-06:
Yeah. In 0.5.x there is no way out, other than disconnecting the gateway for a while. For 0.6 I will separate the gateway stats and avoid this behaviour.

gotthardp, 2018-08-17:
Hello @danielknox, I partially fixed this bug, however the problem is conceptual: the server is updating the gateway location based on GW-provided information. At least in my situation the GPS location changes slightly with every report, so if your gateway provides location info, you could still get this error.

Alternatively I could introduce a check-box "auto-update location" which could disable this behaviour.

gotthardp, 2018-08-23:
Feel free to re-open if needed.

closed 2018-08-23 by gotthardp

danielknox, 2018-09-01:
Hi thanks for this, as GPS will jitter perhaps a percentage change threshold should be introduced before the location is updated behind the scenes? I believe TTNmapper just says 100meters change then assume the gateway was moved, or we could go just with GPS standard error?