gotthardp/lorawan-server #458: Unknown frame: MHDR = 0xC0

JoobyPM, 2018-08-13:
Hi,
A few days ago I found that I've many unknown frames in debug.log.
I've checked them:
C0 28 00 00 FF 80 22 00 01 18 0D CB 3F 08 D1 50 1C 99  <<192,40,0,0,255,128,34,0,1,24,13,203,63,8,209,80,28,153,91>>
C0 28 00 00 FF 85 1E 00 03 20 07 00 00 14 58 21 D1 F3  <<192,40,0,0,255,133,30,0,3,32,7,0,0,20,88,33,209,243,209>>
As I undertand the 0xC0 corresponds to Mtype (Rejoin-request).

- Distribution type - Binary 0.5.5
- Hardware type - VPS
- OS type and version - Linux vps-26862.vps-default-host.net 4.4.0-130-generic #156-Ubuntu SMP Thu Jun 14 08:53:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
- Erlang/OTP version - Erlang/OTP 21 [erts-10.0] [source] [64-bit] [smp:1:1] [ds:1:1:10] [async-threads:1]; Eshell V10.0  (abort with ^G)

gotthardp, 2018-08-13:
Hello. Thanks for a perfect report. I think that C0 corresponds to MType = 110, which is RFU, i.e. "reserved for further update". I think it's an invalid frame.

gotthardp, 2018-08-13:
Hello. Thanks for the perfect report. Yes, the 0xC0 corresponds to Rejoin-Request. This is LoRaWAN 1.1, which is (unfortunately) not supported by this server for now.