gotthardp/lorawan-server #459: Secure gateway communication

brammahh, 2018-08-13:
We're currently exploring the possibility of running a LoRa application server in the cloud instead of locally.  This awesome implementation fits our needs perfectly because of its fault-tolerance and scalability. Because the gateway and server communicate over UDP, security however is a major concern when running this in the cloud.

The LoRaWAN server already supports backend communication over MQTT, but communication with gateways seems UDP-only. To achieve end-to-end security, MQTTS would make a great fit for communication with the gateway. 

Instead of sending to and receiving from `192.168.1.10:1680` over UDP, MQTT topics `lora/downlink/<macAddr>` and `lora/uplink/<macAddr>` could be used. This way communication would be fairly streamlined with the UDP alternative. I assume this would, however, require quite some changes, as the server needs to keep track of the protocol each gateway uses, config files need new options etc. 

Do you see this as a feature that should make it into release, or would we need to implement this ourselves in a fork? How would one go about implementing this without prior Erlang experience? 

Thanks in advance!

gotthardp, 2018-08-13:
But you will need to modify the gateway too. One option would be to use the TTN protocol as requested also by the issue #444. The simplest option is however to use IPsec. Did you consider this option?

brammahh, 2018-08-14:
At first glance, IPsec might seem like the simplest option. When scaling the thing, however, IPsec becomes a major pain, because an IPsec concentrator demands a lot of resources. IPSec is a bit of an overkill for only a few packets UDP packets per minute.

For now, while still exploring the cloud approach, we'll simply write an external application that maps MAC addresses over MQTT to a local port on the cloud server. 

A packet on `lora/downlink/ABCDEF0123456789` for example gets mapped to `localhost:20000`. Any packets coming from LoRaWAN on `localhost:2000` are then relayed back to `lora/uplink/ABCDEF0123456789`. A different port would be allocated for every new gateway. This approach would be fairly lightweight even when many gateways are connected.

If we decide to actually go through with moving the LoRa application server to the cloud, we'll consider implementing this in Erlang, but for now we'll keep it simple.

brammahh, 2018-08-15:
The packet_forwarder is written in C, but with a little effort I think I could manage to update it. I'm not exactly sure what you mean by UDP/TLS though. Isn't TLS most commonly used on top of TCP?

brammahh, 2018-08-18:
We've been considering DTLS as you proposed, but it simply does not make a lot of sense.

Thanks to LoRa's secure nature, the problem does not lie within the encryption of the packets. When someone eavesdrops on communication between the LoRaWAN server and a gateway, the actual LoRa payload remains encrypted. DTLS would prevent man-in-the-middle attacks thanks to certificates, but it would still allow anyone to communicate with the server.

To securely run this a publicly accessible server, authentication is more important than encryption. Even though it won't be very useful, anyone on the internet can pretend to be a certain gateway and send packets over UDP. 

Implementing MQTT as medium for gateway communication would provide both the authentication and encrypt the whole packet. That's by the way exactly [what these guys did](https://www.loraserver.io/lora-gateway-bridge/overview/) in their open-source implementation. They wrote a gateway-side UDP to MQTT proxy which communicates with their application server.

gotthardp, 2018-08-18:
Hello. That is true, but only when client certificates are not used. If you deploy a client certificate to each gateway, you can achieve even a client/gateway authentication using plain DTLS. Client certificates are not an option?
I am aware of the MQTT bridge, but I still believe that using a messaging protocol (like MQTT) for point-to-point connectivity is an overkill. It creates yet another software component to manage and it removes the gateway authorization rules from the server authority-- it's the MQTT broker who decides which gateway can connect.

romansoft, 2018-10-22:
I would also like to see secure out-of-the-box connectivity between gateways and network server. Most of deployments I see run across public internet, so securing this connection is imperative.

VPN is an option, but it's cumbersome to set up. MQTT is easy to set up and is pretty much supported on any server platform and on most gateways.

How about this approach? https://www.loraserver.io/lora-gateway-bridge/install/gateway/multitech/

gotthardp, 2019-05-23:
Since v0.7.0 the LNS over ws/wss will be supported: https://doc.sm.tc/station/tcproto.html

closed 2019-05-23 by gotthardp