gotthardp/lorawan-server #460: Forwarding uplinks

NicoHell1, 2018-08-14:
Hello,
I had a question about a way to forward uplinks received by the network to an application using API (Orange Live Objects if you know), I already implemented an external application which uses network's API to retrieve the uplinks, builds the JSON object corresponding to Live Objects format and sends them to this platform using the provided API but I was wondering if there was not any better solution.
I read carefully the part of the documentation on connectors & handlers but I do not think it will work as I would like to. Indeed, to use this API, I need to specify the API in the HTTP request and the body format is predefined and must stay the same in order to publish the data to this platform. I can give you an example of this format :

NicoHell1, 2018-08-14:
Yeah that is exactly what I did, an external application in Java. I was indeed asking if there was a simpler way and I did not think it was possible even using the parse uplink but maybe I was wrong, you think it is possible this way, building such a JSON with the "Parse Uplink" field ?

gotthardp, 2018-08-14:
It shall be possible directly in the "Parse Uplink" field. You can build any JSON there by specifying a possible nested "Dictionary" in the Erlang syntax. The input Fields are a dictionary too, co you can just put them into the value field. Specify `data` as your Uplink Field and then try something like:
```erlang
fun(Fields, Data) ->
  #{metadata => #{source => <<"urn:lora:{deveui}">>},
    model => <<"model_jri_loraspy_t0_v1">>,
    tags => [],
    value => Fields}
end.
```

gotthardp, 2018-08-14:
If you succeed integrating this with the Orange API, would you mind sharing your fun() here? I would put it in the documentation, so others know how this can be done.

NicoHell1, 2018-08-14:
I will try that, thank you, and yes of course I think I can share it with you. The Orange API uses an API-KEY to allow publishing data, what option of the network should I use to specify the latter ? The Passwork/Key field ?

gotthardp, 2018-08-14:
Oops, this is not possible. It was already requested in the issue #292.

I can implement it quickly, but you will have to use the latest 0.6.0 master branch.

NicoHell1, 2018-08-14:
Ah ok, I missed this issue, I was searching for this topic. Well no problem, I will wait the next release

gotthardp, 2018-08-14:
I just implemented a Header+Token authentication for the future version 0.6.0. The release may take few weeks as I am waiting for a new MQTT client, so you may try compiling the server from sources.

NicoHell1, 2018-08-22:
Hello again, so as you suggested I am currently trying to compile directly from sources on Debian, unfortunately I am facing some issues during compilation, I followed the installation instructions (the manual part) but when I try to run the "make release" command, it fails with the message "Uncaught error in rebar_core". I join the crashdump file, maybe there's something I did wrong
[rebar3.txt](https://github.com/gotthardp/lorawan-server/files/2310470/rebar3.txt)

gotthardp, 2018-08-22:
Please try `apt-get install erlang-parsetools`

NicoHell1, 2018-08-22:
I did, it seemed to work a bit better, however, I have an issue with npm I guess, I get the error "fatal: unable to connect to github.com" with errno = Connection refused, I tried to run npm install -g npm and to add github.com to known hosts but the error keeps appearing anyway

NicoHell1, 2018-08-22:
Ok no problem, I changed the git config --global, just ran the command "git config --global url.https://github.com/.insteadOf git://github.com/"
The problem was probably due to network security restrictions (blocked ports by admin), this allowed me to bypass the blocked outgoing ports I guess

NicoHell1, 2018-08-23:
Back again at the Erlang function to build the JSON to be sent, I assumed in Erlang I could retrieve the actual deveui of the device by using {deveui} in the Parse Upling function (just like in the Connector part where I can use for example api/{deveui}) but maybe I'm wrong, it seems that the HTTP request is not even executed (I tried it with requestbin to inspect the HTTP request but no request has been received).

gotthardp, 2018-08-23:
What exactly did you do? It shall work, but perhaps there is a mistake in the syntax.

NicoHell1, 2018-08-23:
I think that is the problem indeed, I do not know Erlang very well, here's the syntax I used : 

fun(Fields, Data) ->
  #{metadata => #{source => <<"urn:lora:{deveui}">>},
    model => <<"model_jri_loraspy_t0_v1">>,
    streamId => <<"urn:lora:{deveui}!uplink">>,
    tags => [],
    timestamp => {datetime},
    value => Fields}
end.

I was not sure about the "{datetime}" & "{deveui}" parts, can I use this syntax to get these actual values ? And I did not quite understand the use of the "Data" argument

gotthardp, 2018-08-23:
"Data" is the message payload. If you did specify `data` as you "Uplink Field" then you have a Field `data` that contains the same value. So in your case you can safely ignore the "Data" argument.

The "{template}" substitution is not done here, but you can do it manually. If you add `deveui` and `datetime` as your "Uplink Fields" you can extract the value of the Field and then insert it where you want. Like this:
```erlang
fun(#{deveui:=DevEUI, datetime:=DateTime}=Fields, Data) ->
    #{metadata => #{source => <<"urn:lora:", DevEUI/binary>>},
        model => <<"model_jri_loraspy_t0_v1">>,
        streamId => <<"urn:lora:", DevEUI/binary, "!uplink">>,
        tags => [],
        timestamp => DateTime,
        value => Fields}
end.
```

NicoHell1, 2018-08-23:
It seems to work better indeed, the HTTP request is executed, I just get the 401 error code so unauthorized. I have to work on that, it might be related to something else.

gotthardp, 2018-08-23:
Are you able to intercept the HTTP request and verify the body and authorization headers are correct?

NicoHell1, 2018-08-23:
That's what I'm trying to do with requestbin & tests with Postman to compare with a request I know works

NicoHell1, 2018-08-24:
Is there a way to send a list of JSON with the Erlang function ? In order to be accepted, the body of the request has to be " [My_JSON_Here] " but for now I only send the JSON part

gotthardp, 2018-08-24:
Until now, the list of JSON was understood as multiple messages to send.

I made an update, so please do `git pull` and rebuild. With this update if you send list in a list like `[[My_JSON_Here]]` it will be understood as a list of JSONs.

NicoHell1, 2018-08-24:
@lorawan_admin:build:122 Supervisor lorawan_connector_sup had child <<"Live Objects Connector">> started with lorawan_connector_http:start_link({connector,<<"Live Objects Connector">>,<<"TACOS">>,<<"json">>,<<"http://requestbin.fullcontact.co...">>,...}) at <0.740.0> exit with reason no function clause matching lorawan_admin:build([#{metadata => #{source => <<"urn:lora:333934366E367F03">>},model => <<"model_jri_loraspy_t0_v1">>,...}]) line 122 in context child_terminated

That is the debug log when I tried the format [[My_JSON_Here]] unfortunately

gotthardp, 2018-08-24:
I didn't test it, sorry. Please do another update and another try.
Thanks for your patience.

NicoHell1, 2018-08-24:
Sorry for my late answer, I was not there, I just tested it with requestbin, it works for the lists of JSON, now I test it with Live Objects and keeps you informed

NicoHell1, 2018-08-24:
Unfortunately I am still not able to forward data to Live Objects, error 401 so unauthorized, my API-Key is correct so I cannot see why for now, I am trying to compare the headers from a request made from Postman I know works with the request made by the server.
It might be the "Cloudfront-Forwarded-Proto" one which is set to "https" for the working request and set to "http" for the network server one but I am not sure

NicoHell1, 2018-08-24:
Back at it again, it works now, I changed the value of API-Key, maybe it was wrong or this key was deprecated

gotthardp, 2018-08-24:
So, problem solved? You managed to publish to Life Objects?

Would you mind summarizing what needs to be done (and your Fun handler) so I can put it in the documentation, please?

NicoHell1, 2018-08-24:
Yeah! It seems to work pretty well, I ask my manager if I am allowed to share it, I do not think it will be a problem but as I am just a trainee, I have to ask :sweat_smile:

gotthardp, 2018-08-24:
Thank you! I am glad it works. Of course, all private and confidential information will be anonymized.

But I missed your question above: If `data` has to be `payload`, then take the `data` field, push its content into the `payload` field and clear the original `data`:
```erlang
fun(#{deveui:=DevEUI, datetime:=DateTime, data:=Data}=Fields, Data) ->
    #{metadata => #{source => <<"urn:lora:", (lorawan_utils:binary_to_hex(DevEUI))/binary>>},
        model => <<"model_jri_loraspy_t0_v1">>,
        streamId => <<"urn:lora:", DevEUI/binary, "!uplink">>,
        tags => [],
        timestamp => DateTime,
        value => Fields#{payload => Data, data => undefined}}
end.
```

NicoHell1, 2018-08-24:
Thank you for your help, I keep you informed for the possibility to share what has be done.

Yes, I figured out how to get it done just like I wanted so it is ok now I guess :smile:

alireza2980, 2018-08-27:
I don't know what's wrong with what I do but I can not get anything out of LoraWan Server , and I don't get any error or debug message . as I read the documentation it's not mandatory to have a Parse Uplink fun , whatever Fun I try , I get a parse uplink error due to function clause. It's a long time I'm trying to get data out of the software to requestbin , webhook  , or even a local python http server , no success.

NicoHell1, 2018-08-27:
What fields did you select in "Uplink fields"? As far as I know, some fields seem to make the parsing function fail, I think that all the rxq metadata make the function fail (so "rssi", "lsnr", "all_gw" and "bst_gw").
I am not quite sure about that but it seems that when these are selected, the function fails and no request is executed, hope it will help

gotthardp, 2018-08-27:
@NicoHell1, negative: I just tried a Parse Uplink fun with all fields and it works.

NicoHell1, 2018-08-27:
@gotthardp I will test it tomorrow then, maybe I have this because of the function I wrote

NicoHell1, 2018-08-28:
It seems that I have indeed a problem with my Fun expression, in the debug log, I have the message "<PID> exit with reason {badmap, ...} "

gotthardp, 2018-08-28:
And what did you write? :)

NicoHell1, 2018-08-28:
```erlang
fun(#{deveui := DevEui, datetime := DateTime, data := DaTa}=Fields, Data) ->
  [
  [
  #{metadata => #source =><<"urn:lora:", (lorawan_utils:binary_to_hex(DevEui))/binary>>},
     model => <<"model_jri_loraspy_t0_v1">>,
     streamId => <<"urn:lora:", (lorawan_utils:binary_to_hex(DevEui))/binary, "!uplink">>,
     tags => [],
     timestamp => DateTime,
     value => Fields#{payload => lorawan_utils:binary_to_hex(DaTa)}
}
]
]
end.
```

gotthardp, 2018-08-28:
At least one typo. This `#{metadata => #source` should be `#{metadata => #{source`.

NicoHell1, 2018-08-28:
Actually I made a mistake in copying it, the bracket is here in my fun expression

gotthardp, 2018-08-28:
oops. I just made a fix that should fix the error you are getting

NicoHell1, 2018-08-28:
It seems to work perfectly now, and you have now the fun expression I used :) But actually, you can remove "data := DaTa" in the arguments and change the last line from 
```erlang
value => Fields#{payload => lorawan_utils:binary_to_hex(DaTa)}
```
to 
```erlang
value => Fields#{payload => lorawan_utils:binary_to_hex(Data)}
```

closed 2018-08-28 by gotthardp