gotthardp/lorawan-server #463: MQTT Connectivity

KVX-2018, 2018-08-15:
Goog day,
I try to use MQTT connecton to server for forwarding data from Gateway to own server.
Don't see recieving data on server. I think there is not connecting to MQTT broker. Broker based on Node-RED, this is tested with several Broker clients, these work.
Please tell me MQTT protocol works? Please check what i missed. 

LOG
{mqtt,<<"MQTT">>}
2018-08-15 10:35:30.614 [debug] <0.13926.0>@lorawan_connector_mqtt:terminate:249 Connector MQTT terminated: shutdown
2018-08-15 10:35:30.614 [debug] <0.291.0>@lorawan_connector_sup:start_child:16 Start {mqtt,<<"MQTT">>}
2018-08-15 10:35:30.615 [debug] <0.290.0> Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({connector,<<"MQTT">>,<<"MQTT_FWD">>,<<"json">>,<<"mqtt://127.0.0.1:1883">>,<<"out/{devaddr}">>,...}) at pid <0.14081.0>
2018-08-15 10:35:34.316 [debug] <0.14085.0>@lorawan_mac_commands:build_fopts:74 03F31283 <- [dev_status_req]
2018-08-15 10:35:34.316 [error] <0.14085.0> node 03F31283 {not_semtech_mote,{4,<<0,17,17,17,34,34,34>>}}

Screenshot of setting below
[image]

[image]

gotthardp, 2018-08-15:
Hello. I believe you set the Application of the device to "Semtech Mote". You need to set it (in the Device Profile) to the one you defined, i.e. "MQTT_FWD".

KVX-2018, 2018-08-15:
Thanks a lot.  this works

gotthardp, 2018-08-15:
Cool :)

closed 2018-08-15 by gotthardp