gotthardp/lorawan-server #464: JSON package

KVX-2018, 2018-08-15:
Good day.
Set forward filed  Data  only on LORA server, the own Server recieves "data":"333334","text":"334"}.
From Lora node i send data: 333334. What does mean filed text":"334? how possible switch off it ?
BR/Alex

gotthardp, 2018-08-15:
Hello. This is a funny coincidence :) 0x33 is ASCII character "3", 0x34 is "3", so 0x333334 is "334" ASCII.

gotthardp, 2018-08-15:
How to switch it off? Unset (remove) the "Payload" setting to ASCII in your Handler.

KVX-2018, 2018-08-15:
I understand that 0x333334 and "334" ASCII. are connected.
Does the text field in JSON turn off or not?
Does it send by LORA server?

KVX-2018, 2018-08-15:
Thanks, got it.

KVX-2018, 2018-08-15:
maybe add empty field to drop-down list ?

gotthardp, 2018-08-15:
The drop-down list can be unset/cleared.