gotthardp/lorawan-server #469: No logs created

gmbankar37, 2018-08-22:
hello , 
After updating to latest release 0.5.5 I am not able to see any logs or more correctly they are not created. And how much space does logs take as the host is nearly out of space.

gmbankar37, 2018-08-23:
> Hello. In worst case the system creates 2 logs, each of 5 files, 10MB
> each, so upto 100 MB can be eaten by logs before they get rotated. To
> reduce the amount you can
>
>    - reduce the number of logs, e.g. by disabling the debug log
>    - decreasing the maximum file size and decreasing the number of files
>    kept (see https://github.com/erlang-lager/lager#internal-log-rotation)

The problem for now is not reduction but that they are not starting at all.
Is there a way I can delete the received frames?

gotthardp, 2018-08-23:
You can delete all rxframes via a console:
 * start the server
 * connect to the cluster via remote shell by `erl -sname test -remsh lorawan@<hostname>`, where hostname is what you can see in the web-admin
 * delete the database table by entering a command `mnesia:delete_table(rxframes)`.
 * restart (stop and start] the server to recreate the database

gotthardp, 2018-08-25:
And version 0.6.0 will have a "Purge" button to delete all rxframes (and events).

closed 2018-08-25 by gotthardp