gotthardp/lorawan-server #473: When I try to forward uplinks to requestbin I got the following error in error.log (actually I'm very happy at last I got the error message)

alireza2980, 2018-08-27:
Before submitting a bug report please
 - read the Troubleshooting documentation;
 - make sure you are running the latest stable release;
 - inspect the debug.log to check if there are some warnings or errors reported.

gotthardp, 2018-08-27:
This error is related to the LoRaWAN gateway. What kind of gateway are you using? Do you see your uplinks in the Received Frames?
Apparently, if the server cannot receive the uplink, it cannot forward it to the backend.

alireza2980, 2018-08-27:
I use a Kerlink iFemtocell indoor gateway , yes I recieve a lot of uplinks and they are correct , I can manually decode them (they are correct) 
[image]

gotthardp, 2018-08-27:
Could you share your Handler and Connector configuration, please?

alireza2980, 2018-08-27:
[image]
[image]

alireza2980, 2018-08-27:
just a quick note , I got another 1Gate (RPI based) , there I run a 0.5.5 version , it gives absolutely no error when nothing is being sent through the http forwarder.

although the payload is correct , I got two other errors on it : 

2018-08-27 18:57:23.991 [error] <0.631.0>@lorawan_mac:ingest_data_frame:92 <<"1095B3E2">> Unexpected mtype 3
2018-08-27 19:03:15.560 [error] <0.750.0>@lorawan_mac:ingest_data_frame:92 <<"EB020017">> Unexpected mtype 3

I don't know what it means but the payload is correct.
I am on EU868 band.

alireza2980, 2018-08-27:
about the URI , I tried the IP:Port by setting up a python http listener , on webhook and requestbin , it also sends nothing , I checked firewalls,etc.

There is ASCII and CAYENNe , I know my payload is not Cayenne. I really need help ,

gotthardp, 2018-08-27:
If your payload is neither ASCII nor Cayenne, then leave the field empty. Please, read the documentation for Connectors and Handlers, then fix your configuration accordingly and send me updated screenshots for double-checking.

alireza2980, 2018-08-28:
No success , (changing the URI to http://localhost:8062) and changing the payload to "NOTHING" , yet receiving the error , I recompiled the source code but still the problem persists.

2018-08-28 21:33:40.971 [debug] <0.118.0>@lager_handler_watcher:126 Lager installed handler {lager_file_backend,"error.log"} into lager_event
2018-08-28 21:33:40.971 [debug] <0.123.0>@lager_handler_watcher:126 Lager installed handler error_logger_lager_h into error_logger
2018-08-28 21:33:41.428 [debug] <0.113.0>@lager_handler_watcher:126 Lager installed handler lager_backend_throttle into lager_event
2018-08-28 21:33:41.896 [warning] <0.302.0>@lorawan_app:retrieve_valid_ssl:45 http_admin_listen_ssl not configured
2018-08-28 21:33:41.925 [info] <0.334.0> server lorawan@debian started
2018-08-28 21:33:41.962 [debug] <0.342.0>@lorawan_http_registry:update_routes:62 New routes []
2018-08-28 21:33:41.985 [debug] <0.345.0>@lorawan_connector_sup:start_child:16 Start <<"Http">>
2018-08-28 21:33:41.986 [debug] <0.347.0>@lorawan_connector_http:ensure_gun:148 Connecting Http to http://localhost:8062
2018-08-28 21:33:51.768 [info] <0.339.0> gateway 7276FF00390307F8 {connected,{{192,168,1,5},48190}}
2018-08-28 21:35:05.108 [info] <0.370.0> device 244E7B0002000452 {join,<<"017A3D7C">>}
2018-08-28 21:35:05.109 [debug] <0.370.0>@lorawan_mac:encode_accept:414 Join-Accept <<"017A3D7C">>, netid <<0,0,0>>, cflist undefined, rx1droff 0, rx2dr 5, appkey <<"66C9807207D91D1607EA6F8796D122BE">>, appnce <<"EE49F5">>
2018-08-28 21:35:05.109 [debug] <0.370.0>@lorawan_handler:join:112 Join-Accept in RX1: {rxq,868.1,<<"SF12BW125">>,<<"4/5">>,{{2018,8,28},{17,5,4.0}},undefined,1355297388,-58,7.0}
2018-08-28 21:35:09.846 [error] <0.340.0>@lorawan_gw_forwarder:handle_info:108 Ignored PUSH_DATA: JSON syntax error: {'EXIT',{badarg,[{jsx_decoder,value,4,[{file,[47,104,111,109,101,47,116,114,97,118,105,115,47,98,117,105,108,100,47,103$
2018-08-28 21:35:21.973 [debug] <0.371.0>@lorawan_mac_commands:store_actual_adr:143 ADR indicator set to 1
2018-08-28 21:35:21.974 [debug] <0.371.0>@lorawan_mac_commands:build_fopts:77 017A3D7C <- [dev_status_req]
2018-08-28 21:35:22.716 [error] <0.340.0>@lorawan_gw_forwarder:handle_info:108 Ignored PUSH_DATA: JSON syntax error: {'EXIT',{badarg,[{jsx_decoder,value,4,[{file,[47,104,111,109,101,47,116,114,97,118,105,115,47,98,117,105,108,100,47,103$

alireza2980, 2018-08-28:
I am running debian 9 , can I ask you which linux distro the software works best on ? I'm running the server on a VM

alireza2980, 2018-08-28:
on another version I got the debug message :  2018-08-28 18:04:51.142 [debug] <0.327.0>@lorawan_handler:retransmit:217 EB020017 retransmitting

gotthardp, 2018-08-28:
Any distro works. I am worried about this error
```
2018-08-28 21:35:09.846 [error] <0.340.0>@lorawan_gw_forwarder:handle_info:108 Ignored PUSH_DATA: JSON syntax error: {'EXIT',{badarg,[{jsx_decoder,value,4,[{file,[47,104,111,109,101,47,116,114,97,118,105,115,47,98,117,105,108,100,47,103$
```
That's gateway related. Would you be able to find out what did the gateway actually sent? Either from the gateway's log or using wireshark.

gotthardp, 2018-09-07:
Any update? Shall I close the issue?

alireza2980, 2018-09-07:
on vacation , give some days.

closed 2018-10-03 by gotthardp