gotthardp/lorawan-server #474: websocket connect failed

ly1243667342, 2018-08-30:
I am using the software in version(0.6.0). I can receive data from node and view the data from the website(http://192.168.0.108:8080/admin#/rxframes/list).
[image]

but i can not connect to server by websocket.

the log is below:
ws.html:56 WebSocket connection to 'ws://192.168.0.108:8080/ws/nodes/00513128' failed: Error during WebSocket handshake: Unexpected response code: 404
connect @ ws.html:56
toggle_connection @ ws.html:73
onclick @ ws.html:144
ws.html:56 WebSocket connection to 'ws://192.168.0.108:8080/ws/nodes/00513128' failed: Error during WebSocket handshake: Unexpected response code: 404

what is wrong? thanks !

gotthardp, 2018-08-30:
How did you configure the connector?

ly1243667342, 2018-08-31:
Thank you for your timely reply.
this is the configure of connector.
[image]

now i can connect to server .but can not receive data via websocket. website(rxframes/list) can recieve data normally.
[image]

what is wrong ?

gotthardp, 2018-08-31:
You don't have Application set. You need to have a Handler of some name and that (Application) Name has to be set both in the Connector as well as in the Device Profile your devices are using.

ly1243667342, 2018-09-06:
i don know how to set application. i try to set application,but not work!
[image]
[image]
[image]

the websocket still can not receive data, website can receive data and show data. can you give a workable configure? thank you !

gotthardp, 2018-09-06:
Set Publish Uplinks to `/ws/uplinks/{devaddr}`. You have to use the template, not the real DevAddr.

gotthardp, 2018-09-18:
Any update? Shall I close the issue?

ly1243667342, 2018-09-22:
thank you very much

closed 2018-09-22 by ly1243667342