gotthardp/lorawan-server #475: Wrong output data format

digitra, 2018-08-31:
Hello!
I started to use lorawan-server version 0.4.13 (for Debian) with MQTT type connectior to external application.
The data recieved from devices seemed to be correct and was like following example:
iotadmin@LORA-NET:~$ mosquitto_sub -t 'out/#' -v -u iotclient1 -P xxxxxx 
out/03BD9274 {"data":"0306001D61086515547598000700000000020200000008000000000900000000D204000000","datetime":"2018-08-08T14:53:22Z","devaddr":"03BD9274","deveui":"001BC50350000113","fcnt":36,"gateway":{"mac":"0000E8EB1141908A"},"group":"IoT_group","port":2,"rxq":{"codr":"4/5","datr":"SF12BW125","freq":869.1,"lsnr":7.2,"rssi":-78,"tmst":4290822428}}

But after I got lorawan-server upgraded to version 0.5.5 ( actually just deinstalled V0.4.13 and installed new version ) something happened to
the data format. It became like following:
iotadmin@LORA-NET:~$ mosquitto_sub -t 'out/#' -v -u iotclient2 -P orange
out/{devaddr} {"text":"\u0003\u0006\u0000\u001d\u0018\b?\u001bf?\u0000\u0007\u0000\u0000\u0000\u0000\u0002\u0002\u0000\u0000\u0000\b\u0000\u0000\u0000\u0000\t\u0000\u0000\u0000\u0000?\u0004\u0000\u0000\u0000"}

All setting in handler/connector for the old and new version are the same.
I tried also version 0.5.3 and 0.5.4, but with the same result. Even docker conterised docker installation of the lates lorawan-server demonstrates the same behavior.

Could you give as a tip what could be the couse of such strange output data format?
Thanks in advance
Boris

gotthardp, 2018-08-31:
Hello. You likely set Handler Payload to "ASCII Text", but the payload wasnt an ASCII text. Clear the Payload field and the strange `text` will disappear.

digitra, 2018-09-03:
Hello, Clearing "Payload" field in the "Handlers" tab did help.
Now device data are readable.
Thanks a lot.
Boris

gotthardp, 2018-09-03:
Cool. I am glad it works.

closed 2018-09-03 by gotthardp