gotthardp/lorawan-server #476: I use the latest version of the server(0.6.0), not receiving uplink data.

welt1, 2018-09-04:
I've used 0.4.13, 0.5.4, and 0.5.5 versions of server, which are good to use. 

Today, I used the latest version of the server, but found no uplink data, only the downstream packets from server, and the Dashboard window's icon turns blue, as shown in the following figure: 

[image]
[image]
[image]

From the gateway log, my uplink packets send success,  and received the Ack
from the server.
[image]

My settings for server are consistent with the old version, is there anything that needs to be changed in this version? 

Please help to see, where is the problem? Thank you!

gotthardp, 2018-09-04:
Did you define some Groups and Areas, reviewed the Gateway and Profile configuration and assigned those to some Group/Area? If yes, please have a look into the error.log of the server.

welt1, 2018-09-04:
yes, i defined.
I used the original version of 0.60 in July, no problem, the setting is similar to now, also need to define groups and Areas.
error.log only one log:
2018-09-04 11:15:18.211 [error] <0.362.0> node 023CE114 <<"too_early">>

gotthardp, 2018-09-04:
<<"too_early">> is a downlink related error. Does the server really print nothing upon uplink? The cyan timeline boxes you see are downlinks only (I suppose Class C, right?) There must be a reason for not seeing any uplinks.

gotthardp, 2018-09-04:
And the LED changes or not? Does the server work and doesn't log, or doesn't even work? But if it sends downlinks then it has to work.

welt1, 2018-09-04:
Looks all right, just no upstream packet, here is the Debug.log record:

2018-09-04 17:39:15.174 [debug] <0.1432.0>@lorawan_application_semtech_mote:handle_rxq:35 PUSH_DATA <<2,60,225,20>> 4369 0 240
2018-09-04 17:39:27.371 [debug] <0.1435.0>@lorawan_application_semtech_mote:handle_rxq:35 PUSH_DATA <<2,60,225,20>> 4369 0 240
2018-09-04 17:39:36.617 [debug] <0.1438.0>@lorawan_application_semtech_mote:handle_rxq:35 PUSH_DATA <<2,60,225,20>> 4369 0 240
2018-09-04 17:39:46.673 [debug] <0.1441.0>@lorawan_application_semtech_mote:handle_rxq:35 PUSH_DATA <<2,60,225,20>> 4369 0 240
2018-09-04 17:39:58.019 [debug] <0.1446.0>@lorawan_application_semtech_mote:handle_rxq:35 PUSH_DATA <<2,60,225,20>> 4369 0 240
2018-09-04 17:40:03.917 [debug] <0.1449.0>@lorawan_application_semtech_mote:handle_rxq:35 PUSH_DATA <<2,60,225,20>> 4369 0 240

gotthardp, 2018-09-04:
But that *is* an upstream packet. It seems that only the uplinks are not logged.

welt1, 2018-09-04:
I also found,  when I send a data format wrong upstream package.
This uplink package will be recorded.
[image]
[image]
 Debug.log record:
2018-09-04 18:20:14.642 [error] <0.1792.0> node 023CE114 {not_semtech_mote,{2,<<17,17,0,0,17,17,240,0,0,17,0,0,34,34,34>>}}

gotthardp, 2018-09-04:
It is really weird. On my installation I can see all uplinks from the mote.

gotthardp, 2018-09-04:
Could you please "Purge" the frames you have, let the mote sent few frames and then display the frames via `http://localhost:8080/api/rxframes`.

welt1, 2018-09-05:
Thank you for your help!
I  cleared the original frames, and re-tested it. Let mote send 7 frames, the first 6 correct format data, the last one is the format wrong data. The result is only the last record of the upstream frames.
[image]

[image]

http://127.0.0.1:8080/api/rxframes?_sortField=datetime&_sortDir=ASC 
[image]

[image]

welt1, 2018-09-05:
[image]

From this view, it is true that the server received 7 upstream data, sending 6 downstream data (the last error format did not). But why is the uplink data not recorded?

welt1, 2018-09-05:
[image]

This is the rxframe record in Mnesia. From this view, it seems that the server received uplink data at the same time send the downlink data, if there is downlink data to ignore uplink data, only record the downlink data, no downlink data to record the uplink data.

gotthardp, 2018-09-05:
You are very skilled with mnesia :)
What is the machine / OS you are using? Is it something special? Could you run `erl` and then type `erlang:system_info(os_system_time_source).` and let me know what is the outcome?

I suspect you have insufficient time precision and the frames overwrite each other. Could it be possible on your system?

welt1, 2018-09-05:
PS F:\work\erlang\temp\lorawan-server> erl
Eshell V9.2  (abort with ^G)
1> erlang:system_info(os_system_time_source).
[{function,'GetSystemTime'},
 {resolution,100},
 {parallel,yes},
 {time,1573058956161024}]
2>

Would that be the problem?

gotthardp, 2018-09-06:
Yes, I think that's the problem. Your resolution is too small. The "frame id" is generated based on system_time and your is moving too slowly (because of the low resolution) so two frames (uplink and downlink) get the same ID and the later (downlink) overrides the former (uplink).

Just for comparison, I am getting
```
1> erlang:system_info(os_system_time_source).
[{function,clock_gettime},
 {clock_id,'CLOCK_REALTIME'},
 {resolution,1000000000},
 {parallel,yes},
 {time,1536228835314423705}]
```

Your resolution is small. And I also suspect the time will have always 1024 at the end, which means the last byte is always zero.

gotthardp, 2018-09-06:
Would you do one more test for me, please?

Run the following command:
```erlang
{os:timestamp(), erlang:system_time(millisecond), erlang:system_time(microsecond), erlang:system_time(nanosecond)}.
```

4 times, very quicky after each other (as quickly as you can :) and post here what numbers you got. I consider this as a bug, but need to find a way how to safely generate the ID.

welt1, 2018-09-07:
Oh, there's a problem, you're right. Here are the test results:

[image]

gotthardp, 2018-09-07:
Thanks for the tests. I made an update, which should fix this problem. This will be released as 0.6.0 in the coming days.

welt1, 2018-09-10:
Testing the new version is OK, this problem has been solved, thank you very much!