gotthardp/lorawan-server #478: [Feature Request] Hide keys by default

cmenscher, 2018-09-04:
Anywhere NwkSKey and AppSKey are viewable/editable in a form, they should appear hidden at first. 
 That way, absent minded screenshotters won't accidentally share the keys for a device to an entire user support mailing list, not to mention prevent casual onlookers.  😣

gotthardp, 2018-09-05:
I agree with the request, but my programming skills are not sufficient to implement it.

altishchenko, 2018-09-05:
And my belief is that no untrained person should get *any* access to the NS interface. For them is the App, that filters everything and uses the REST API to modify the server's configuration.

GeorgSteyn, 2019-09-30:
Hi, I am looking for places to contribute to so I saw this issue.

I do understand both arguments and I please want to know should this change be implemented or not.

If this should not be implemented, could this issue be closed, please?

cmenscher, 2019-09-30:
Well in terms of priority I'm not sure it's the highest. But visually obfuscating fields with secure content is pretty standard procedure, even if a simple "show" icon/button is provided to revert to visible.