gotthardp/lorawan-server #481: Search · sys.config

gotthardp, 2018-09-08:


altishchenko, 2018-09-08:
??

gotthardp, 2018-09-09:
This is just a quick remark for me. People searched too much the sys.config. I wanted to improve the documentation to better explain where it is.

altishchenko, 2018-09-09:
Yep. That is why 'I see' :) I was quite for some time thinking along the lines of the more user friendly config in, say, `/etc/lorawan-server/`. Possibly, like gateways do, a bunch of presets with symbolic links to usual JSON (which is more familiar to people, then Erlang) and two defaults storages behind in `sys.config` and `.app`. This way, if you delve into it, yes it will introduce an extra module to load and store all of this stuff in a table of sorts and properly workout the defaults, but then it will open the possibility to introduce mandatory ADR configs depending on the country and frequency plan used. Just imagine, that installing the server you won't have to recite all of them 8 or 16 channels in the EU range configuring a network...

gotthardp, 2018-09-10:
You can always push predefined network config via the REST API (like import). With this issue I only wanted to create a dedicated Config guidelines, which I just did.

closed 2018-09-10 by gotthardp

altishchenko, 2018-09-12:
Sorry. I think I misunderstood your intentions with this issue note. No bother. Pushing data via REST is ok, but it is not a 'boxed' solution, because for this you are required to have an extra app that will push the data in. It is not as "choose some defaults and then work your way up". The way it is right now is appropriate for the clusters, but it is a bit of a headache for 'multiple client separate instances' cloud configuration, where you have a bunch of different clients and anyone of them has to have a complete configured server instance all for themselves, just no devices registered - this is for the app to do.
What we do right now is - we package the server along with the app software _including_ the preset mnesia database, that contains a number of preset configuration settings (network/ADR/handler/connector/etc included). So that real app just manages the addition of gateways and nodes in a user friendly way. (All them numbers are on the QR coded stickers on the devices, so you only need to copy-paste the digits into the app's form). No diving into the technology and details of the radio frequency plans...