gotthardp/lorawan-server #483: websocket new connector

gmbankar37, 2018-09-11:
Hii, 
As the websocket path for uplinks is defined like /ws/uplinks/{Dev_addr} so when my dev_addr changes do I have to create a new websocket connector?
In short should i have different connector for every unique device (node) ? Or can I make it generic?

gotthardp, 2018-09-11:
Well, it is generic. The `{devaddr}` is a template. You shall use this text as is. The replacement is done automatically by the server.

gmbankar37, 2018-09-11:
I have  /ws/uplink/{devaddr} written in publish uplinks and On the demo client I am trying to access ws/uplink/00000AB8. But I am not getting any connection.

gmbankar37, 2018-09-11:
Got it working thank you.

gotthardp, 2018-09-11:
How did you solve the problem, please?

gmbankar37, 2018-09-12:
Oh rookie mistake I just reloaded the server and it started working.

gotthardp, 2018-09-12:
Cool! Thanks.

closed 2018-09-12 by gotthardp