gotthardp/lorawan-server #484: A lot of HTTP errors

b1551, 2018-09-11:
Hi,

I have a lot of HTTP errors: {404,"/admin/rgraph/56540001",{127,0,0,1}} generated every few seconds
[image]

Version: 0.5.5
Distribution type: binary
# uname -a
Linux SilTtn 4.15.0-1014-azure #14~16.04.1-Ubuntu SMP Thu Jun 14 15:42:55 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
npm version 5.6.0
node version 9.10.1
erlang 20

Can someone please help ?

gotthardp, 2018-09-11:
Hello. Did you have had a node "56540001"? Do you still have it? Is there a web-browser opened on your machine, where the administration of this node is opened?

b1551, 2018-09-11:
Yes, I had it, but it's deleted now.
I close the browser and tried again, same thing happens.

uto, 11. ruj 2018. u 15:20 Petr Gotthard <notifications@github.com> napisao
je:

gotthardp, 2018-09-11:
There must be a browser window opened somewhere, which is asking for data about the deleted device.

b1551, 2018-09-11:
Hm, now it stopped. Could be someone else had the window open. Thanks!

uto, 11. ruj 2018. u 15:32 Petr Gotthard <notifications@github.com> napisao
je:

closed 2018-09-11 by gotthardp