gotthardp/lorawan-server #488: Update docker image

myfreescalewebpage, 2018-09-19:
Hello,
I would like to update the docker container with the new one available at "https://hub.docker.com/r/gotthardp/lorawan-server".
What is the manual procedure for that ? The doc indicates how to activate automatic updates, but I want to manage it myself.
I have tried to pull the last image and restart it but I have lost the configuration of the server. I have switch back to a backup so I can try again but I don't find the procedure.
Thanks fro your help.
Joel

gotthardp, 2018-09-19:
Did you follow the standard procedure, i.e. do you use the `--volume /path/to/local:/storage` parameter? If yes, the configuration shall be stored there. Do you see 'mnesia' files there?

myfreescalewebpage, 2018-09-20:
Hi, Thanks for your quick answer.  I think yes, but no way to be sure of course. I have just found the mnesia files in some '/var/lib/docker' directories....  Result of find just below. Is it possible to recover them ?

[find.txt](https://github.com/gotthardp/lorawan-server/files/2401108/find.txt)

gotthardp, 2018-09-20:
This is not the database content. There should be a `Mnesia.lorawan@localhost` diectory with a set of *.DCD and .*DCL files. (instead of localhost there could be the name of your host).

myfreescalewebpage, 2018-09-20:
Thanks for the info. Find 'Mnesia.lorawan@lorawan' (not 'localhost')directory, copied to make a backup and created new container with the latest version of the docker. New installation has 0.6.1 version and my configuration is back :)

closed 2018-09-20 by myfreescalewebpage

gotthardp, 2018-09-20:
Thanks. I am not a Docker expert, but wouldn't `--restart on-failure` be better?

myfreescalewebpage, 2018-09-20:
Haven't tried this option, my understanding is that on-failure is just if the container crashes. 'always' permit to start the container on boot of the host.

gotthardp, 2018-09-20:
On Failure means *Restart only if the container exits with a non-zero exit status. Optionally, limit the number of restart retries the Docker daemon attempts.*
I am worried that with restart always you cannot stop the container intentionally-- it will always get restarted even when you stop it. But users can choose any option they want anyway.

myfreescalewebpage, 2018-09-20:
Sure :)