gotthardp/lorawan-server #490: unable to start again, suddently lorawan_server exited with reason: shutdown

hallard, 2018-09-26:
I played with version 0.6.1 yesterday without any problem while checking #453 and this morning the server is down. I tried to launch it but it shutdown at start and I'm unable to understand why

Here is the debug log file
```
2018-09-26 10:47:27.381 [debug] <0.98.0>@lager_handler_watcher:126 Lager installed handler {lager_file_backend,"error.log"} into lager_event
2018-09-26 10:47:27.384 [debug] <0.100.0>@lager_handler_watcher:126 Lager installed handler error_logger_lager_h into error_logger
2018-09-26 10:47:27.389 [debug] <0.85.0> Supervisor gr_param_sup started gr_param:start_link(gr_lager_default_tracer_params) at pid <0.102.0>
2018-09-26 10:47:27.389 [debug] <0.84.0> Supervisor gr_counter_sup started gr_counter:start_link(gr_lager_default_tracer_counters) at pid <0.103.0>
2018-09-26 10:47:27.390 [debug] <0.86.0> Supervisor gr_manager_sup started gr_manager:start_link(gr_lager_default_tracer_params_mgr, gr_lager_default_tracer_params, []) at pid <0.104.0>
2018-09-26 10:47:27.390 [debug] <0.86.0> Supervisor gr_manager_sup started gr_manager:start_link(gr_lager_default_tracer_counters_mgr, gr_lager_default_tracer_counters, [{input,0},{filter,0},{output,0},{job_input,0},{job_run,0},{job_time,0},{job_error,0}]) at pid <0.105.0>
2018-09-26 10:47:27.518 [info] <0.31.0> Application lager started on node lorawan@ch2i
2018-09-26 10:47:27.529 [debug] <0.110.0> Supervisor mnesia_sup started mnesia_sup:start_event() at pid <0.111.0>
2018-09-26 10:47:27.531 [debug] <0.110.0> Supervisor mnesia_sup started mnesia_ext_sup:start() at pid <0.112.0>
2018-09-26 10:47:27.542 [debug] <0.113.0> Supervisor mnesia_kernel_sup started mnesia_monitor:start() at pid <0.114.0>
2018-09-26 10:47:27.544 [debug] <0.113.0> Supervisor mnesia_kernel_sup started mnesia_subscr:start() at pid <0.115.0>
2018-09-26 10:47:27.549 [debug] <0.113.0> Supervisor mnesia_kernel_sup started mnesia_locker:start() at pid <0.116.0>
2018-09-26 10:47:27.551 [debug] <0.113.0> Supervisor mnesia_kernel_sup started mnesia_recover:start() at pid <0.117.0>
2018-09-26 10:47:27.570 [debug] <0.55.0> Supervisor kernel_safe_sup started dets_sup:start_link() at pid <0.119.0>
2018-09-26 10:47:27.570 [debug] <0.55.0> Supervisor kernel_safe_sup started dets_server:start_link() at pid <0.120.0>
2018-09-26 10:47:27.601 [debug] <0.55.0> Supervisor kernel_safe_sup started disk_log_sup:start_link() at pid <0.124.0>
2018-09-26 10:47:27.601 [debug] <0.55.0> Supervisor kernel_safe_sup started disk_log_server:start_link() at pid <0.125.0>
2018-09-26 10:47:27.655 [debug] <0.113.0> Supervisor mnesia_kernel_sup started mnesia_tm:start() at pid <0.118.0>
2018-09-26 10:47:27.657 [debug] <0.113.0> Supervisor mnesia_kernel_sup started mnesia_checkpoint_sup:start() at pid <0.139.0>
2018-09-26 10:47:27.659 [debug] <0.113.0> Supervisor mnesia_kernel_sup started mnesia_snmp_sup:start() at pid <0.140.0>
2018-09-26 10:47:27.659 [debug] <0.113.0> Supervisor mnesia_kernel_sup started mnesia_controller:start() at pid <0.141.0>
2018-09-26 10:47:27.664 [debug] <0.113.0> Supervisor mnesia_kernel_sup started mnesia_late_loader:start() at pid <0.142.0>
2018-09-26 10:47:27.664 [debug] <0.110.0> Supervisor mnesia_sup started mnesia_kernel_sup:start() at pid <0.113.0>
2018-09-26 10:47:27.664 [info] <0.31.0> Application mnesia started on node lorawan@ch2i
2018-09-26 10:47:27.704 [debug] <0.162.0> Supervisor eid_sup started eid_server:start_link() at pid <0.165.0>
2018-09-26 10:47:27.704 [info] <0.31.0> Application eid started on node lorawan@ch2i
2018-09-26 10:47:27.744 [info] <0.31.0> Application crypto started on node lorawan@ch2i
2018-09-26 10:47:27.762 [info] <0.31.0> Application cowlib started on node lorawan@ch2i
2018-09-26 10:47:27.804 [info] <0.31.0> Application asn1 started on node lorawan@ch2i
2018-09-26 10:47:27.804 [info] <0.31.0> Application public_key started on node lorawan@ch2i
2018-09-26 10:47:27.821 [debug] <0.252.0> Supervisor ssl_sup started ssl_manager:start_link([]) at pid <0.253.0>
2018-09-26 10:47:27.823 [debug] <0.252.0> Supervisor ssl_sup started tls_connection_sup:start_link() at pid <0.254.0>
2018-09-26 10:47:27.824 [debug] <0.252.0> Supervisor ssl_sup started ssl_listen_tracker_sup:start_link() at pid <0.255.0>
2018-09-26 10:47:27.825 [debug] <0.252.0> Supervisor ssl_sup started dtls_connection_sup:start_link() at pid <0.256.0>
2018-09-26 10:47:27.827 [debug] <0.252.0> Supervisor ssl_sup started dtls_udp_sup:start_link() at pid <0.257.0>
2018-09-26 10:47:27.827 [info] <0.31.0> Application ssl started on node lorawan@ch2i
2018-09-26 10:47:27.828 [debug] <0.261.0> Supervisor ranch_sup started ranch_server:start_link() at pid <0.262.0>
2018-09-26 10:47:27.829 [info] <0.31.0> Application ranch started on node lorawan@ch2i
2018-09-26 10:47:27.831 [debug] <0.266.0> Supervisor cowboy_sup started cowboy_clock:start_link() at pid <0.267.0>
2018-09-26 10:47:27.831 [info] <0.31.0> Application cowboy started on node lorawan@ch2i
2018-09-26 10:47:27.834 [info] <0.31.0> Application gun started on node lorawan@ch2i
2018-09-26 10:47:27.835 [info] <0.31.0> Application jsx started on node lorawan@ch2i
2018-09-26 10:47:27.836 [info] <0.31.0> Application iso8601 started on node lorawan@ch2i
2018-09-26 10:47:27.837 [info] <0.31.0> Application gen_smtp started on node lorawan@ch2i
2018-09-26 10:47:27.840 [info] <0.31.0> Application xmerl started on node lorawan@ch2i
2018-09-26 10:47:27.841 [info] <0.31.0> Application recon started on node lorawan@ch2i
2018-09-26 10:47:27.841 [info] <0.31.0> Application rabbit_common started on node lorawan@ch2i
2018-09-26 10:47:27.845 [info] <0.31.0> Application amqp_client started on node lorawan@ch2i
2018-09-26 10:47:27.847 [info] <0.31.0> Application emqttc started on node lorawan@ch2i
2018-09-26 10:47:27.851 [debug] <0.286.0> Supervisor mongodb_supervisor started mongodb:start() at pid <0.287.0>
2018-09-26 10:47:27.853 [info] <0.31.0> Application erlmongo started on node lorawan@ch2i
2018-09-26 10:47:27.856 [warning] <0.290.0>@lorawan_app:retrieve_valid_ssl:45 http_admin_listen_ssl not configured
2018-09-26 10:47:27.862 [debug] <0.309.0> Supervisor {<0.309.0>,ranch_listener_sup} started ranch_conns_sup:start_link(http, supervisor, 5000, ranch_tcp, 5000, cowboy_clear) at pid <0.310.0>
2018-09-26 10:47:27.864 [info] <0.322.0> server lorawan@ch2i started
2018-09-26 10:47:27.876 [debug] <0.94.0>@lager_handler_watcher:126 Lager installed handler lager_backend_throttle into lager_event
2018-09-26 10:47:27.877 [debug] <0.330.0>@lorawan_http_registry:update_routes:62 New routes []
2018-09-26 10:47:27.885 [debug] <0.330.0>@lorawan_http_registry:update_routes:62 New routes [{<<"/pylight/up">>,lorawan_connector_ws,[{connector,<<"pylight-ws">>,<<"ch2i-pylight">>,<<"json">>,<<"ws:">>,undefined,<<"/pylight/up">>,<<"/pylight/events">>,undefined,undefined,undefined,true,undefined,undefined,undefined,undefined,undefined,undefined,undefined,[],0,0,undefined},uplink]},{<<"/pylight/events">>,lorawan_connector_ws,[{connector,<<"pylight-ws">>,<<"ch2i-pylight">>,<<"json">>,<<"ws:">>,undefined,<<"/pylight/up">>,<<"/pylight/events">>,undefined,undefined,undefined,true,undefined,undefined,undefined,undefined,undefined,undefined,undefined,[],0,0,undefined},event]}]
2018-09-26 10:47:27.887 [debug] <0.333.0>@lorawan_connector_sup:start_child:16 Start {mqtt,<<"pylight-mqtt">>}
2018-09-26 10:47:27.889 [debug] <0.335.0>@lorawan_connector_mqtt:connection_args:128 Connecting pylight-mqtt to <<"mqtt://localhost:1883">>, id undefined, user <<"ch2i">>
2018-09-26 10:47:28.001 [warning] <0.30.0> lager_error_logger_h dropped 34 messages in the last second that exceeded the limit of 50 messages/sec
2018-09-26 10:47:28.305 [debug] <0.340.0>@lorawan_connector_ws:websocket_init:95 WebSocket connector <<"pylight-ws">> with #{}
2018-09-26 10:47:28.305 [debug] <0.341.0>@lorawan_connector_ws:websocket_init:95 WebSocket connector <<"pylight-ws">> with #{}
2018-09-26 10:47:28.305 [debug] <0.342.0>@lorawan_connector_ws:websocket_init:95 WebSocket connector <<"pylight-ws">> with #{}
2018-09-26 10:47:28.610 [info] <0.327.0> gateway 0004B827EB9C6093 {connected,{{86,66,32,161},42104}}
2018-09-26 10:47:28.871 [warning] <0.324.0>@lorawan_db_guard:send_alert:196 gateway FCC23DFFFE0F44B5 [{<<"disconnected">>,100}] []
2018-09-26 10:47:28.877 [error] <0.324.0>@lorawan_db:-foreach_record/3-fun-1-:276 gen_server <0.324.0> terminated with reason: no match of right hand value {aborted,{missing_from,[{mimemail,check_headers,2,[{file,"/home/travis/build/gotth [... cut ...] ,"/hom..."},...]},...]}} in lorawan_db:'-foreach_record/3-fun-1-'/3 line 276 in context child_terminated
2018-09-26 10:47:30.919 [error] <0.322.0> Supervisor {<0.322.0>,lorawan_sup} had child db_guard started with lorawan_db_guard:start_link() at <0.348.0> exit with reason reached_max_restart_intensity in context shutdown
2018-09-26 10:47:30.920 [info] <0.31.0> Application lorawan_server exited with reason: shutdown
```
Any idea what's going on, yesterday when I left all was working fine.
thanks

gotthardp, 2018-09-26:
It is trying to send an e-mail, but you didn't specify a From header, so it crashed.
```
terminated with reason: no match of right hand value {aborted,{missing_from,[{mimemail,check_headers,2
```

gotthardp, 2018-09-26:
I just made a fix for this. Are you able to build it from sources?

gotthardp, 2018-09-26:
Concerning the `dpkg -i`: That's also a bug (issue #491). I have fixed that also, so since 0.6.2 the system will track sys.config modifications and warn before overriding. Thanks for pointing this out.