gotthardp/lorawan-server #492: Server logging issue

gmbankar37, 2018-10-01:
I am using server version 0.5.5 and still the logs are not getting created. Also I am using raspberrypi as the host and it is almost out of memory. Is this the reason the logs are not being created? 
[image]

gotthardp, 2018-10-01:
Insufficient RAM is not likely the reason why logs are not being created. Who is owner of the `/var/lib/lorawan-server` directory?

gmbankar37, 2018-10-01:
The user is pi. In this account only the server is running.

gmbankar37, 2018-10-01:
I mean its not root.

gotthardp, 2018-10-01:
Unless you changed something by default the service is running under the user `lorawan`.
see https://github.com/gotthardp/lorawan-server/blob/master/scripts/lorawan-server.service

gmbankar37, 2018-10-01:
How do I delete the server and reinstall it entirely? not just updating.

gotthardp, 2018-10-01:
via dpkg or manually? dpkg has a command for that `sudo dpkg -r lorawan-server`. They you may doublecheck the /user/lib/lorawan-server and /var/lib/lorawan-server and /var/log/lorawan-server are deleted

gmbankar37, 2018-10-01:
Where do I find the lorawan-server.service file?

gotthardp, 2018-10-01:
`/lib/systemd/system`

gmbankar37, 2018-10-03:
what does packet_collision means? After how much time does it resends the collided packet?

gotthardp, 2018-10-03:
Packet collision is a gateway issue, see https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT#L425

The collided packet is lost, but standard LoRaWAN functions for confirmed transmission apply.

gmbankar37, 2018-10-04:
After how much does it resents the collided frame? Does the server has provision to avoid collision?

gmbankar37, 2018-10-04:
If I deleted the server and installed it with previous mnesia files would it work ?

gotthardp, 2018-10-04:
What exactly is the problem you are experiencing? Is the collision between class A devices or is there a Class C involved? If it's class A and you have only one gateway the server cannot do anything.

gotthardp, 2018-10-04:
Is it from Class A or Class C devices? The downlink slots are scheduled by the device. If two devices schedule a downlink slot for the same time the server cannot do much about it.

gmbankar37, 2018-10-04:
It is a class A device.

gotthardp, 2018-10-04:
So, if two devices send an uplink at the same time, the collision happens and one downlink is lost. If it was a confirmed downlink it will be retransmitted once the device sends another uplink. The server cannot do much about it.

gmbankar37, 2018-10-05:
Do you how to increase the channels in IN865 band? I mean I cannot randomly assign frequencies from 865 to 867.

gotthardp, 2018-10-05:
Google search gave me this article that lists more frequencies: http://www.ensembletech.in/lora-frequency-bands-india/
Try setting those for your Network config.

gmbankar37, 2018-10-09:
Yes I chose from those frequencies only.

gotthardp, 2018-10-09:
Good. Does this solve your issue?

gmbankar37, 2018-10-10:
To some extend it solves packet collision. I have one question though. the
scenario is
 Consider 2 nodes and one channel (866.1 MHz). device 1  transmit one
packet and then it opens two receive window (Rx1 and Rx2). If it gets
downlink in Rx2 in same channel and device 2 transmits from same channel
when the Rx2 is up.
In this case does collision happen? and if yes how to solve this problem?

gotthardp, 2018-10-10:
In that case there is no collision. The collision happens when two devices open their RX window at the same time.

closed 2018-11-22 by gotthardp