gotthardp/lorawan-server #493: Connector status in list does not match status in connector's Status tab

zumbrunnen, 2018-10-02:
I am currently investigating an issue with one of our HTTPS/REST connectors. The corresponding entries in `debug.log` look like this:

```
2018-10-02 10:09:51.913 [debug] <0.867.0>@lorawan_connector_http:handle_info:139 Unknown message: {gun_up,<0.868.0>,http2}
2018-10-02 10:09:57.372 [debug] <0.330.0>@lorawan_connector_http:handle_info:139 Unknown message: {gun_up,<0.331.0>,http2}
2018-10-02 10:10:10.021 [debug] <0.867.0>@lorawan_connector_http:ensure_connected:182 my-connector-name failed to connect: timeout
2018-10-02 10:10:10.021 [warning] <0.867.0>@lorawan_connector_http:handle_info:70 Connector <<"my-connector-name">> not connected, uplink lost
2018-10-02 10:10:11.926 [debug] <0.867.0>@lorawan_connector_http:handle_info:139 Unknown message: {gun_up,<0.868.0>,http2}
2018-10-02 10:10:17.385 [debug] <0.330.0>@lorawan_connector_http:handle_info:139 Unknown message: {gun_up,<0.331.0>,http2}
```

For short, the LoRa server cannot/does not send the uplink JSON requests, although the application works just fine... but this is a different topic.

The status tab of the connector (/admin#/connectors/edit/my-connector-name) displays an amber triangle for this connection. On the other hand, the list view of all connectors (/admin#/connectors/list) shows a green checkmark for this connector.

**Connector List**
[image]

**Connector Details**
[image]

IMHO the list's icon/status should be the same as in the details...

gotthardp, 2018-10-02:
Yeah, this is a known deficiency/bug, which is not easy to solve, unfortunately. But I know this is bad and need to be fixed.

BTW, do you know that placing mouse cursor above the triangle will show some more details about the connection?

zumbrunnen, 2018-10-02:
Hi @gotthardp 
Thank you for your fast reply. Yeah, I know about the hovering over the icon - it just says "Connecting".