gotthardp/lorawan-server #494: Support HTTP/2 for Connectors

zumbrunnen, 2018-10-02:
Related to my log output in #493 we found the cause of problem with the connector not working. We observe a `gun_up`, meaning that the connection is up, but the problem seems to be with the selection/use  of the underlying gun implementation (see also: `gun_http.erl` and `gun_http2.erl` in https://github.com/ninenines/gun/tree/master/src)

This is triggered when we have a normal HTTP/1.1 connection: https://github.com/gotthardp/lorawan-server/blob/16bf722d67299c9f44ad0b4127c50eb4153833a5/src/lorawan_connector_http.erl#L86-L87

And this is triggered when we get a HTTP/2 connection (via ALNP): https://github.com/gotthardp/lorawan-server/blob/16bf722d67299c9f44ad0b4127c50eb4153833a5/src/lorawan_connector_http.erl#L138-L140

To us, this seems like a bug as it shouldn't matter if we get a HTTP/1.1 or a HTTP/2 connection from the server the uplink shall be sent to, especially because the gun library seems to support and use http2.

gotthardp, 2018-10-02:
Thank you for such a great bug report. You are exactly right. In fact, the protocol is ignored for gun_down, but is not ignored for gun_up. It's probably an copy-paste error.

gotthardp, 2018-10-02:
I just fixed the `{gun_up, C, http}` issue. Would you be able to build the latest revision from sources?

zumbrunnen, 2018-10-02:
Yes, sure. We are using the Docker containers, so I'm going to build a fresh one and try it. Will report back in an hour or so (as I first have to revert the changes on the Kubernetes Ingress to disable http2 for the moment :wink:)

zumbrunnen, 2018-10-02:
Hey @gotthardp, I just built the image from master (it says version 0.6.2) and the connector is coming up! It connects successfully with HTTP/2! Thank you very much! :+1:

gotthardp, 2018-10-02:
Cool! Problem solved :) Thanks for your help.

closed 2018-10-02 by gotthardp