gotthardp/lorawan-server #495: Dashboard timeline meaning of different packets received

gmbankar37, 2018-10-05:
Can you tell what is the meaning of all these coloured packets received in the screenshot attached. Thank you. 
[image]

gotthardp, 2018-10-05:
Good question. I just updated the documentation to say

 * Rolling timeline that displays:
   - **red** error events;
   - **yellow** warning events;
   - **blue** handled uplinks frames;
   - **cyan** sent downlinks frames;
   - **gray** ignored uplinks frames.

gmbankar37, 2018-10-08:
So what do you mean by ignored uplinks frame? I my case after with every blue frame it shows a gray frame so I thought it might be downlink. So according to my screenshot why are there so many ignored frames?

gotthardp, 2018-10-08:
Share with me the Frames list. The items of the dashboard are taken from there.

gotthardp, 2018-10-08:
You forgot the attachment

gmbankar37, 2018-10-08:
I did attach it. It is an image.

gotthardp, 2018-10-08:
But I don't see any image :)

gmbankar37, 2018-10-08:
[image]
So what are the unconfirmed downlink frames with no data?

gotthardp, 2018-10-08:
It looks like a downlink from the server MAC (without application data). Could you please hover the mouse cursor above the arrow on the first line and let me know what text it displays?

gmbankar37, 2018-10-09:
It displays down. 

[image]

gmbankar37, 2018-10-09:
What is the condition where we get missed uplink warning?

gotthardp, 2018-10-09:
Missed uplink is generated when there is a gap in the FCnt sequence.

gmbankar37, 2018-10-10:
So that frame is not transmitted with that particular FCnt. Say FCnt=30 is transmitted and received and then the next FCnt received is 33. So can i say 31 and 32 are not transmitted or is it just that they are missed by server?

gotthardp, 2018-10-10:
The missing frames are either not sent or not received. You cannot say more without some RF level debugging.

gmbankar37, 2018-10-10:
And what would be the reason for missing any frames?

gotthardp, 2018-10-10:
Usually a RF interference with the environment or other transmissions on the same frequency.

gmbankar37, 2018-10-10:
Also the channels(i.e. frequencies) i have defined at the node does not
match with the frequencies received. I mean i have defined channel
frequency 866.400 but at the same channel I receive from different
frequency.

gotthardp, 2018-10-11:
This depends on the gateway.

closed 2019-03-16 by gotthardp