gotthardp/lorawan-server #496: Using MongoDB authentification

Heiner92, 2018-10-05:
I set up authentification for my MongoDB and as a result can't get my connector to publish the received uplinks any more (it used to work just fine before setting up auth).

I tried playing with the authentification options of the connector but nothing seems to work.
Is is at this point possible to authenticate to a MongoDB?
If not would you consider adding this as a feature in the near future? If so, I would also need to specify the authentification DB as is possible in MongoDB Compass (screenshot below).

[image]

gotthardp, 2018-10-05:
Hello. MongoDB authentication *should* be possible since version 0.6.0. It's a feature of an external library I use, so I didn't test this feature thoroughly. If it doesn't work I can try raising an issue.

The authentication source however feels not supported by the [mongodb library I use](https://github.com/SergejJurecko/erlmongo).

Heiner92, 2018-10-06:
I can't find anything about authentification in the library you linked. Could you please show me how you used it? I hope to find a way of making this work with a custom auth DB (usually you just have to connect to the auth DB first, send user & pw and then you can connect to all DBs the user has rights to access)

This is how you can do it in python.

```server='servername'
port=27017
username='username'
password='mypassword1234'
authSource='auth'
authMechanism='SCRAM-SHA-1'
uri = "mongodb://{}:{}@{}:{}/?authSource={}&authMechanism={}".format(username,password,server,port,authSource,authMechanism)
mongoclient = MongoClient(uri)```

gotthardp, 2018-10-06:
In case you can read Erlang, the implementation is here https://github.com/SergejJurecko/erlmongo/commit/4d0c1ee4efd3deab6e256bac78bb71cd316be6d2

I just supply Username and Password as two parameters. Do you know the MongoDB protocol? It not easy to find how the auth db is actually used in the protocol.