gotthardp/lorawan-server #498: Configuration of a nano-gateway

BouTayehGaby, 2018-10-09:
In my configuration file of the nano-gateway (Pycom expansion board equipped with a Lop4) I have the following :
SERVER = 'http://localhost:8080'
PORT = 1680
In the main code of the nano-gateway, I open a UDP socket with the server as follows:

self.server_ip = usocket.getaddrinfo(self.server, self.port)[0][-1]
self._log('Opening UDP socket to {} ({}) port {}...', self.server, self.server_ip[0], self.server_ip[1])
self.sock = usocket.socket(usocket.AF_INET, usocket.SOCK_DGRAM, usocket.IPPROTO_UDP)
self.sock.setsockopt(usocket.SOL_SOCKET, usocket.SO_REUSEADDR, 1)
self.sock.setblocking(False)

NB: the server and port variables are the ones defined in the configuration file.

My nano-gateway connects without any problem with the local router and opens a UDP socket but it is not able to communicate with the server at the address: 'http://localhost:8080'.
The gateway is shown as "Disconnected". 

am I missing something?

BouTayehGaby, 2018-10-09:
Hello,
yes it sends a pull and push request after a prefixed period : 

self.stat_alarm = Timer.Alarm(handler=lambda t: self._push_data(self._make_stat_packet()), s=60, periodic=True)
self.pull_alarm = Timer.Alarm(handler=lambda u: self._pull_data(), s=25, periodic=True)

It worked just fine with "The things network" (TTN) I was able to send data to TTN without any problem.

gotthardp, 2018-10-09:
Not sure what could be wrong. Is the server address correct? Above you wrote
```
SERVER = 'http://localhost:8080'
```
which is definitely wrong. I guess it should be
```
SERVER = 'localhost'
```

BouTayehGaby, 2018-10-10:
I have tried it but I still have the same problem.

gotthardp, 2018-10-13:
To debug this further I'd need a https://www.wireshark.org trace of the gateway to server communication.

BouTayehGaby, 2018-10-19:
Hello again,

I am not an expert in Networking so I am gonna explain what I have done so far.
1- I have a nano-gateway that consists of a Pycom expansion board and a Lopy4.
2- The Network server is however installed on my Windows PC.
3- The nano-gateway gateway and the PC are connected to the same local router.

The gateway opens a UDP socket:
self.sock = usocket.socket(usocket.AF_INET, usocket.SOCK_DGRAM, usocket.IPPROTO_UDP)
self.sock.setsockopt(usocket.SOL_SOCKET, usocket.SO_REUSEADDR, 1)
self.sock.setblocking(False)

And once it receives a packet from a sensor it sends it via the socket:
self.sock.sendto(packet, (self.IP,self.port))

If I am not wrong, the self.IP cannot be "localhost" it must be the private IP of the PC where I have installed the Network server!
and the port should be "1680".

So instead of having self.IP = "localhost", I initialized the IP to the private IP address of my pc.
So in theory, when the gateway receives a packet it must transmit it to my pc and to the port "1680" where the network server installed on the pc is listening to upcoming packets.

I have installed Wireshark as you have suggested and I am getting the following multiple times: 
source : IntelCor_c4:cd:81	
Destination: Broadcast	
Protocol: ARP
Length: 42	
Info: Who has 192.168.4.1? Tell 192.168.4.2
knowing that 192.168.4.1 is the IP of the Lopy4 (nano-gateway).

And this is appearing multiple times too:
Source: 192.168.4.2	
Destination: 239.255.255.250	
Protocol: SSDP
Length: 216	
Info: M-SEARCH * HTTP/1.1 

Apparently, the PC is not able to recognize the nano-gateway(192.168.4.1)? and it is sending ARP packets?

BouTayehGaby, 2018-10-22:
No there is no UDP packets to port 1680.

gotthardp, 2018-10-22:
And can you ping the machine? (`ping IP`, where `IP` is the address if the machine where the server is running)

BouTayehGaby, 2018-11-16:
I have ordered the 1-Gate LoRaWan gateway, they have this network server already installed.
It worked just fine now. I have a few questions regarding the functioning of the network server though, I will start another thread for these questions since it's a different topic.

Thank you for your help!

BouTayehGaby, 2018-11-16:
Yes, it is solved you can close it.

closed 2018-11-16 by gotthardp